XCSoar/XCSoar #610: Untranslated strings on splash screen / prompt window

scls19fr, 2021-05-05:
XCSoar versions having and not having the problem
-------------------------------------------------

XCSoar 7.6

Expected behavior
-----------------

Strings on splash screen should be translated.

"Quit" button
"What do you want to do?"

Actual behavior
---------------

[image]

Steps to reproduce the behavior
-------------------------------

Set language to a non english (Config / System / Look / Language, Input).
Quit XCSoar
Run XCSoar

Do you have any idea what may have caused this?
-----------------------------------------------

https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/src/Dialogs/SimulatorPromptWindow.cpp#L112

uses  `_` macro not `N_` macro

same for

https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/src/Dialogs/SimulatorPromptWindow.cpp#L60

Do you have an idea how to solve the issue?
-------------------------------------------

Using `N_` macro instead.

But such changes may not be enough (because I don't know XCSoar startup process enough)

MaxKellermann, 2021-05-28:
When that screen is displayed, XCSoar hasn't loaded any profile yet. Without a profile, your profile's language configuration is not set.

scls19fr, 2021-05-28:
Maybe profile should be loaded before this page is shown?

MaxKellermann, 2021-05-28:
Which profile? Profile selection comes after that.

scls19fr, 2021-05-28:
I didn't saw any option to choose a specific profile because I was only using default profile. Maybe choosing profile should simply be done before showing this page.
An other option could be to use default system language until profile is not loaded

lordfolken, 2021-11-09:
XCSoar uses the  system locale to define a language on initial start. So we could also use this to change the start screen. If the user then loads a different language in his profile, that's a conscious choice on his behalf.

closed 2026-03-18 by lordfolken