XCSoar/XCSoar #611: Kobo Clara HD not anymore supported (touchscreen not working)

wolkenjaeger, 2021-05-06:
XCSoar versions having and not having the problem
-------------------------------------------------

6.8.17

Expected behavior
-----------------

Touchscreen inputs should work

Actual behavior
---------------

Touchscreen inputs are not working. However, the physical button works to turn Kobo on and off.

Steps to reproduce the behavior
-------------------------------

Install Kobo 7.6 on a Kobo Clara HD

Do you have any idea what may have caused this?
-----------------------------------------------

Kobo Clara HD is not defined in the source code. 

Do you have an idea how to solve the issue?
-------------------------------------------

Add relevant definitions in the source code, maybe it is just a driver issue during boot. I am able to help with basic testing.

MaxKellermann, 2021-05-06:
This can be analyzed only if a developer has a Kobo Clara HD.

lpojzl, 2021-08-21:
@MaxKellermann: Max, I bought Kobo ClaraHD just for XCsoar. As it's currently not working, I am open for a 1/2 year loaner to you. Are you willing to have a look at the ClaraHD related issues? It would be great having it working before next season. ;-)
The device would come flashed with latest Kobo software and XCsoar build installed, together with non-intrusive serial2usb cable for console debugging.
[image]

sepi, 2021-08-24:
I also have the same issue on the Clara HD with the the 7.x versions. Will see if I can debug somehow. For now, version 6.8.11 works ok.

fagat73, 2021-11-08:
Hello,
Any news regarding this issue ? I've installer the latest XCSoar 7.20 version on my Kobo Clara HD and the touch screen not working anymore.
Thank you.

MaxKellermann, 2021-11-08:
@fagat73 of course not. If there were news, you would be able to read them here. No developer has a Kobo Clara HD, and there cannot ever be any progress until a developer has one.

fagat73, 2021-11-08:
I've got the Kobo but I'm not a developer on the projet. Anyway let me know if I can be of any help with resolving this issue.

lordfolken, 2021-11-08:
You can organize a crowd funding for a device for @MaxKellermann that way the project can support the platform. Without hardware it will be very difficult.

MaxKellermann, 2021-11-08:
You can help by either
- becoming a developer and solve the issue, or
- donating a Kobo Clara HD to the XCSoar project

Just owning a Kobo Clara HD is not helpful at all, if you're not using it to analyze the problem and submit the fix to this project.

fagat73, 2021-11-08:
OK let me know how i can donate to the XCSoar project and i'll do

MaxKellermann, 2021-11-08:
My address can be found on the [XCSoar imprint](https://xcsoar.org/imprint.html). If one day a Kobo Clara HD spawns at my doorstep, I'll take care for this issue.

wolkenjaeger, 2021-11-17:
@fagat73 did you send a device to @MaxKellermann ? If not, I can do it.

fagat73, 2021-11-17:
Hi. Not yet. I’m looking for a second hand one. And it’s not yet Christmas :) if you got one please do. And thank you.

wolkenjaeger, 2021-11-17:
I will do so.

commit 3d1a59d1 (2021-12-03): Fix startup for Clara

This adds a 0.1s delay in rcS file for Kobo to allow touch screen
and display to initialize properly.  this was specifically tested
for Clara HD (N249) but other models have had similar problems and
those models may now work better.

Closes #611
Closes #317
not sure if #515 is fixed because I don't have that model of Kobo
closed 2021-12-03 by MaxKellermann