XCSoar/XCSoar #614: 15 infoboxes split in 3 rows

scls19fr, 2021-05-07:
Hello,

smartphones have screen which are bigger and bigger...
it could be interesting to use screen size increase to display more data (for people who want such setup).
A "15 infoboxes split in 3 rows" layout could be an interesting layout setup.
(especially when / if #613 is implemented)

Kind regards

fronzee, 2021-05-25:
Would be perfect to have more infoboxes. Now I have to use multiple pages, just to have several more infoboxes. 

Or to allow "dragging" of infoboxes to a wanted place and customising each.

MaxKellermann, 2021-09-03:
There's already a 8x3 layout, which is landscape only; I'll implement a portrait 8x3 layout.
5x3 doesn't leave enough space for the map, it's unusable.

closed 2021-09-03 by MaxKellermann

scls19fr, 2021-09-04:
Thanks @MaxKellermann I will try this layout... even if I think it's a lot of infoboxes (8x3=24).
The counterpart being that they may be quite small

You are saying that "5x3 layout doesn't leave enough space for the map, it's unusable"... 

Hower I'm not sure 5x3 layout leave less space for the map than my current layout which is 4x3=12 (12 infoboxes split in 3 rows)

Maybe 6x3=18 or 7x3=21 should also be considered.

(I will proabably try to tackle this when I will have some free time)

My opinion about layout, is that layout could be defined differently on each page.

A page full of big infoboxes with no map (or a very small map) should be a possible configuration but an other page with bigger map and less infoboxes could also be used.

I think it should be possible to define a default infoboxes layout... each page should use by default this layout when no specific layout is given to a page. The harder part of this will be how to manage settings of these layouts according both mode (circling, cruise, ...) and page.

elgandoz, 2021-09-06:
I second the idea of 5x3 layout, maybe with a couple of position variants.

@MaxKellermann please consider re-opening this issue.

Recent phones have a high aspect ratio, very "tall and narrow", something around 20∶9 ([detailed list here](https://en.wikipedia.org/wiki/Comparison_of_high-definition_smartphone_displays)): such ratio would make 5x3 layout a great compromise between amount of information and readability.

Here's a comparison of the `10 Split`, `12 Split in 3 rows` and a mocked-up `15 Split in 3 rows` respectively, on my _Galaxy S10e_ (2.5 years old phone), which has an AR of 19:9:
[image]

As you can notice, in the latter the available space for the map is actually bigger then `12 Split in 3 rows`. The layout works as well in landscape .

I will try to submit a PR based on the same approach taken in #613.

fronzee, 2021-09-06:
Long awaited feature!

elgandoz, 2021-09-06:
Screenshots from the compiled build.

| [image] | [image] |
|:---:|:---:|

fronzee, 2021-09-06:
> Screenshots from the compiled build.
> [image: Screenshot_20210906-213259_XCSoar.jpg]
> <https://user-images.githubusercontent.com/5859926/132211705-ede7ec3b-f931-40d7-a2da-5b65d9d61183.jpg> [image:
> Screenshot_20210906-213416_XCSoar.jpg]
> <https://user-images.githubusercontent.com/5859926/132211752-92dc248f-e8c7-43a6-a656-87e3040e4277.jpg>

Also if possible even bigger counts like 7x3, 8x3 split the same way! I
would not need to have several pages in this case. I am sure many would
really appreciate. I am using galaxy s9 and there are plenty of space.

scls19fr, 2021-09-06:
8x3 layout (24 infoboxes split in 3 rows) ever exists now both for portrait and landscape thanks to @MaxKellermann commit https://github.com/XCSoar/XCSoar/commit/c9799b2a296ccf43b34a4f77dce0121cdd713200
5x3 layout (15 infoboxes split in 3 rows) may be added by @elgandoz thanks to PR https://github.com/XCSoar/XCSoar/pull/711
so there is a room for
- 6x3 layout (18 infoboxes split in 3 rows)
- and 7x3 layout (21 infoboxes split in 3 rows)

MaxKellermann, 2021-09-06:
We don't need more hard-coded layouts... we need the code to be dynamic, allowing arbitrary rows and columns.

fronzee, 2021-09-06:
8x3 is only in 3 rows together not 1+2 bottom... I tried today 3x8 and its
too much in one place. but 2 at the bottom and 1 at the top would be
better.
Best would be random location for infoboxes like used to be in Winpilot and
now in Seeyou navigator... I am sorry, but I am just a user, not
programmer, cant help.. not in C anyway.

scls19fr, 2021-09-15:
I'm looking for "15 Split in 3 rows" string in https://hosted.weblate.org/projects/xcsoar/translations/fr/ for translation but I couldn't find it. Maybe updating strings on weblate should be triggered by an admin?

scls19fr, 2021-09-15:
Following same idea... the 2 following strings can't be fin in weblate

- "10 Top or Left"
- "24 Bottom or Right"

PR #711 (2021-09-07): InfoBoxes/Layout: add 3x5 InfoBoxes layout
Brief summary of the changes
----------------------------
Adds a new 3x5 layout, aimed at modern mobile phones Aspect Ratios, exactly as done in 202d7f0faff847c74861bd8cc91969f18fb6793c

Related issues and discussions
------------------------------
Closes #614