XCSoar/XCSoar #617: xcsoar 7.6 kobo version - most port options missing

mgiepen, 2021-05-09:
XCSoar versions having and not having the problem
-------------------------------------------------
Problem? 7.6
No problem? 6.16

Expected behavior
-----------------
Usually there were many port options, e.g. ttyACM0 port

Actual behavior
---------------
Port selection limited to "Disabled", "TCP client", "TCP port", "UDP port", "ttymcx0"

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2021-07-02:
Probably since commit 2de78d4aa22cee2913db0031befa0f51019e3d2b
Prior to this commit, all ports were listed, even those that did not exist. Now only ports are listed which actually exist.

What makes you believe that ttyACM0 should exist? Which hardware device should it represent?

mgiepen, 2021-07-02:
Should be the USB Port (USB OTG cable). Would like to use that Port with a XCTracer II Vario. To choose this specific Port on the kobo is described onthe XCTracer Homepage. 
Greetings
Michael

MaxKellermann, 2021-08-28:
I tried to debug this one yesterday, but the microUSB OTG cable (passive) that came with one of my Androids didn't work on the Kobo. Do I need a special OTG adapter for the Kobo?

GrantHeaney, 2021-09-03:
No special OTG cable needed. I use a 4 port USB powered hub, a USB A female to female adaptor and a standard micro USB. It's important to connect it all up and then power the Kobo on, or have the kobo completely off then plug the kobo in so it powers iteslf on.   This connection setup allows the Kobo to have power as well as USB devices connected to it.

MaxKellermann, 2021-09-06:
Doesn't work for me. I have a powered 4 port USB hub as well, and a micro USB to USB A OTG adapter, and a USB-A to USB-B cable to connect OTG adapter with the hub. The Kobo doesn't power itself up when plugging it in, and when I power it up manually, the kernel doesn't detect any USB devices. Sorry, can't help with this problem.

GrantHeaney, 2021-09-07:
That's a bit strange it's not working for you. Connected to my test setup is a USB GPS, a Volkslogger (via USB to serial dongle), and a sound dongle. This is connected to a Kobo Glo HD. 
In my glider i also have vario/airspeed and so on. I did need to configure the default.prf from V6.8.17 for the initial USB configuration and then copy that to the V7.xx Kobo. It actually works really well but if i need to change USB settings i need to go back to V6.8.xx to create a new default.prf.
[image]

closed 2022-06-28 by lordfolken