XCSoar/XCSoar #618: Android 11

n316163, 2021-05-10:
XCSoar is nothing short of brilliant, but the latest Android update has caused a real issue:  Android 11 seems to have restricted xc's access to the GPS sensor, when it runs in the background.

This means it can't produce a track or position unless you keep xc running in the foreground.  Which you can, but..

It stops you using xc just purely as an automatic flight tracker, with the screen off, which was brilliant for logging a day's local flights.  It also loses the gps if you for instance, access the camera app while running xc.

And it's not even XCSoar's fault...

johnthomas22, 2021-05-10:
There's a detailed guide to conforming to the new standard of course. Looks like it's aimed at ensuring users are best enabled to understand what permissions they are granting, and why. 

https://developer.android.com/training/location/permissions#request-location-access-runtime

scls19fr, 2021-11-11:
I noticed 

https://support.google.com/googleplay/android-developer/answer/9799150?hl=en

Will background access location be restored ?

It seems that it have been removed from latest XCSoar version because of Google policy about GPS data

https://github.com/XCSoar/XCSoar/releases/tag/v7.20

lordfolken, 2021-11-12:
V7.20 was released without background GPS as xcsoar was banned from google play with it enabled. 
The project  basically needs to prove why xcsoar needs background location access. 

To this end we need a privacy policy which is now available as privacy.MD
The second thing needed is a video showing the features that need background location access. There is an example in your link. 
Perfect time to get involved.

zinuzoid, 2021-11-12:
https://www.youtube.com/v/YXk9dfTKQRY

This is how I did it in my fork, it passed the store requirement. 

Happy to create a pull request if we want to use the same approach.

lordfolken, 2021-11-12:
@zinuzoid that would be ideal. Please do.

zinuzoid, 2021-11-17:
Is anyone able to help with the wording. I'm not sure what would be the best official text here.

My current permission request dialog text is `XCSoar collects location data to enable live navigation calculation and IGC logger even when the app is in the background` - `Continue`

Please suggests if anyone think we should rewording it.

rawtaz, 2021-11-17:
Perhaps `XCSoar needs to collect location data to enable live navigation calculation and IGC logger, even when the app is in the background` - `Continue` conveys a bit better that this is really needed?

scls19fr, 2022-07-22:
This commit https://github.com/XCSoar/XCSoar/commit/d576049906a5e9e047338ca22c124a4c7ac6552a which was added in v7.24 seems to disable background location ;-(

See https://github.com/XCSoar/XCSoar/releases/tag/v7.24

lordfolken, 2023-09-22:
@MaxKellermann reneabled this in https://github.com/XCSoar/XCSoar/commit/dd05d787db7aba4658dfefb60235976539a4b1f4 for v7.37. Thx Max!

PR #785 (2021-11-26): Android background location prominent disclosure dialog
Closes #618

Here's the video for Google Play Store: https://www.youtube.com/watch?v=h14BWgzDD0w
closed 2023-09-22 by lordfolken