XCSoar/XCSoar #620: Back to waypoints list after watching waypoint detail

scls19fr, 2021-05-10:
Hello,

XCSoar version
--------------

I'm using XCSoar 7.6

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

After seeing waypoints list to watch distance of several waypoints, I want to have more information about some specific waypoints (such as required glide ratio).
So I can choose one and look for data such as required glide ratio or Alt diff MC safety.

Unfortunately when clicking on "close" button... it's getting back to map page... not waypoint lists which is not very convenient.... especially when considering several waypoint options.

Similar problem also occurs when using Nav / Alternates window

Changing in this context "Close" button to a "Back" button... which turn back to waypoints list will be a nice feature to have

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Waypoint details page should store the context of the "previous" page which was used to display waypoint detail.
If it's waypoints list or alternates, close button should be turned to a back button

lordfolken, 2021-05-11:
Please provide a PR.

scls19fr, 2021-05-11:
Sorry @lordfolken but it doesn't seems simple enough to me (or at least some guidance will be required)

I noticed the `close_button` at

https://github.com/XCSoar/XCSoar/blob/c1b2909248a3a45cb1225ce875fce280ac412658/src/Dialogs/Waypoint/dlgWaypointDetails.cpp#L439-L440

but I'm pretty sure some changes are also required on the "caller" side 

https://github.com/XCSoar/XCSoar/blob/c1b2909248a3a45cb1225ce875fce280ac412658/src/Dialogs/Waypoint/WaypointList.cpp#L228-L230

MaxKellermann, 2021-10-14:
It is debatable whether this is a good idea. I don't think it is. If you just want to look up a waypoint and inspect it, you then need an extra click to return to the map.
But in any case, the linked patch is bad because it creates a new list dialog every time. This loses all state.