XCSoar/XCSoar #624: Waypoints list should sort (by default) waypoints according distance even when no filter is set for distance

scls19fr, 2021-05-11:
Hello

XCSoar versions having and not having the problem
-------------------------------------------------

I'm using XCSoar 7.6

Expected behavior
-----------------

Down/Up - Nav - Waypoints list
Name : *
Distance : *
Direction : *
Type : Landable

should show nearest waypoints marked as Landable sorted from the nearest waypoint to the farthest waypoint.

To sort waypoints using distance, a filter need to be applied but filtering and sorting are 2 differents actions.

Actual behavior
---------------

with such filters enabled, waypoints are not sorted using distance (but are displayed using either order of waypoints or alphabetical order... I can't say exactly)

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

Maybe an optional checkbox : "sort by distance" should be set by default.

An other UI idea could be to have an other dropdown
Sort :
- Unsorted
- Alphabetically
- By distance

But it's important that UI can stores last sort choice (or have at least sorted by distance by default)

Kind regards

scls19fr, 2021-08-28:
The problem with alternates list is that it only shows reachable airports (that's the purpose of such a list!)

Aiports which are very near from being reachable but which are not are not displayed in this list.

So an other usefull improvement in waypoints list could be to draw differently airports (like in alternates list) which are reachable from airports which are not (using green circle)

elgandoz, 2021-09-05:
It would be enough set a default distance (like 25Km) instead of the current default "*", with empty results, which is not useful.

I often use this to communicate my distance from a specific waypoint (not a _landable_) via radio, accessing the menu via the "L" gesture: having immediately the closest waypoints, ordered by distance, would save few extra taps.

One good thing is that you need to set the filter once, than it stays during the session, so the problem is present only the first time you open the dialog.

It could also be useful to have a radial/relative bearing from the waypoint, but I'm going to open a different issue about that.