XCSoar/XCSoar #625: not correct interpretation of keyboard input in XCSoar

Blaubart, 2021-05-15:
If I choose NMEA Device at menu, I can select device with up and down, but I can't choose edit, monitor, reconnect etc. using the arrows right and left at the keyboard. I tried XCSoar 7.6 for Linux and OpenVario. At the Windows Version it works.

MaxKellermann, 2021-06-02:
This is designed to work only with devices which are keyboard-only (i.e. no mouse, touch screen or other pointing device). This is the code:
https://github.com/XCSoar/XCSoar/blob/77e12207bee05c5a80000332a27eacbe701b4541/src/Form/ButtonPanel.cpp#L354-L360
On Windows, there is always assumed to be a mouse, so this code is not used (and I cannot confirm that keyboard selection works on Windows - it doesn't here, as designed).
The `HasPointer()` check is only implemented for XCSoar builds targetting the Linux console with libinput.
OpenVario probably qualifies (does it have a touch screen?), but since OpenVario uses a patched XCSoar, I cannot help you - I can only support unpatched, vanilla XCSoar builds.

Blaubart, 2021-06-03:
In my case under Windows 10 with the current XCSoar 7.7 downloaded from your website, the keyboard works perfectly. Here is a video: https://vdrblaubart.dynvpn.de/s/BzaopWKq6F5dE6a

Under Ubuntu 18.04, also XCSoar from your website, the keyboard for selecting the submenu does not work. Here is a video: https://vdrblaubart.dynvpn.de/s/aHryZo3tjenYDtY

What reason exists that the keyboard only has this function on devices that do not use a mouse, touchscreen or similar? Basically, the keyboard could still have the full function.

lordfolken, 2021-06-29:
I think since the bug is present in vanilla xcsoar, it should be considered. You can easily reproduce with keyboard cursor keys.

closed 2022-01-14 by MaxKellermann