XCSoar/XCSoar #629: tcp ports in use

phoebusb, 2021-05-18:
**XCSoar versions having and not having the problem
-------------------------------------------------**

Problem first appeared in 7.0_preview18 and persists through version 7.7
Version 6.8.18 does not have the problem

**Expected behavior
-----------------**

When I connect through WLAN to a Wifi bridge on the ports 4352, 4353, 2000, the port will be connected (in menu "devices" the status is "no data" if no data is sent or whatever data is received "Flarm;GPS" etc. and the data is reflected in the GUI.

**Actual behavior
---------------**

When I connect through WLAN the port is "not connected" and when I press "monitoring" it flashes "port in use" for a short time. No data at all is received.

**Steps to reproduce the behavior
-------------------------------**

It only happens on my Hisense e-ink with Android 10. On the Sony Xperia with Android 10 everything works normally. However, on the sony it will only connect the ports once I am asked "wlan has no internet, do you want to stay connected?" and I press "yes". The hisense doesn't ask at all. 

But 6.8.18 works perfectly on the Hisense, so it's not a major concern for me, but maybe someone else also has the problem as well.

**Do you have any idea what may have caused this?
-----------------------------------------------**

maybe the Android SDK / API version? Also worth to mention that the Hisense comes without google play store and google gimmicks (for known reasons) ...

**Do you have an idea how to solve the issue?
-------------------------------------------**

Unfortunately not ...

MaxKellermann, 2021-06-02:
There is something wrong with this bug report.
> when I press "monitoring" it flashes "port in use" for a short time

This message does not exist in the XCSoar source code. Please verify and describe what exactly you have been doing, cite the exact message. Tiny details which may seem unimportant to you are really important for developers.

phoebusb, 2021-06-03:
Hi Max,
You're right, it's quite strange. But from my collegues no one has that problem, so I guess it's really a special case related to the hisense phone and it's not a major drama. Anyway, I'll test it again tomorrow evening. The flashing message disappears really fast and that's the only hint I got so far besides that it doesn't acquire any data ... I'll try to test it also with the new 7.8 version, maybe it works now ...

phoebusb, 2021-06-04:
Hi,
I just tested with the newest 7.8 code and I got it working. However, I didn't succeed in reproduce the message and It takes quite a while (3-5 min.) until the ports shift from "not connected" to "no data" or "GPS fix etc.". It definitely seems an oddity of the Hisense phone related to newer Android apk/sdk (must probably be, since 6.8 works without any problems) and it's not worth being further investigated ... So I'll close this issue. Thanks for having had a look at it and especially for the 7.8 version fixing the file download problem!

closed 2021-06-04 by phoebusb

MaxKellermann, 2021-06-04:
This was probably fixed by b809bc89ff7407fcbafea5aa8e74d5b8193e4913 or 0d6f2bc733ef0ed2320af3560b99aea9428645e8, but then this bug report was inaccurate and wrong.

phoebusb, 2021-06-06:
sorry to deceive you Max, I did my best, but I'm well aware that the road to hell is paved with good intentions ... For the sake of completeness, I tested again 7.7 version and the behaviour is the same, after 3 - 5 minutes they go to "no data" (when I'm sending nothing), but if I press reconnect the return to "Not connected" ... The only difference I spot is the "device occupied" message in monitoring which only appears in 7.7. but not in 7.8 (see https://github.com/XCSoar/XCSoar/commit/0d6f2bc733ef0ed2320af3560b99aea9428645e8) ...