XCSoar/XCSoar #631: task avg. not being calculated

fronzee, 2021-05-25:
Hi, 

I noticed task avg. is not being shown/calculated, instead shows nothing (not sure if dashed or empty) in the box. 
I am using latest 7.7 version. I did not have such problem in 6.8 last year. 
Now a lot of times crossing start line or having crossed line and selecting next waypoint manually, task avg. and last H avg. shows nothing. Used to work anyway and show speed. Normal message of a start is received with speed, time, altitude, but speed is not calculated.

MaxKellermann, 2021-07-01:
I can't reproduce this problem; whenever I start the task, the TaskAvg InfoBox shows data. Can you give more details on how to reproduce the problem? Is this only with certain tasks? Can you reproduce it in simulator mode with a simple task?

MaxKellermann, 2021-08-16:
No reply.

closed 2021-08-16 by MaxKellermann