XCSoar/XCSoar #632: GR avg calculation

fronzee, 2021-05-26:
All my known versions had it.
newest 7.7, but also earlier versions. version 6 had this problem, too.

Expected behavior
-----------------
GR avg. calculates avg. GR between thermals OK, but if in the same glide you come back to the same place (like going to TP and turning 180 degrees back). GR avg and distance of the glide starts reducing and so the GR avg. 

It should continue adding distance, but it starts to reduce it, therefore GR reduces, too.

sandrinr, 2021-05-26:
Hi fronzee

> GR avg. calculates avg. GR between thermals

Are you sure you are talking about the "GR avg"? What you are describing sounds more like "GR cruise".

"GR avg" just divides traveled distance by altitude difference (over the configured integration period) not taking thermals into account.
What you describe seems to be true for "GR cruise" though. At least I could demonstrate this behavior in a quick simulator run.

You seem to claim this being a bug. However, this probably depends on the definition of "GR cruise".
I assume (but don't know) this is working as intended.

fronzee, 2021-05-27:
Yes, indeed I was talking about GR Cruise. I use it often, behaviour is ok most of the time except when turning a TP. GR avg would not be achieving the same since it would do 3 min avg at most and not all distance/altitude travelled since last thermal.

closed 2021-05-27 by fronzee