XCSoar/XCSoar #641: METAR and TAF not updating on 7.7 and 7.8 on Android

infeeeee, 2021-06-03:
XCSoar versions having and not having the problem
-------------------------------------------------

Tested in the following versions:

Working: 
- 7.7 Windows
- 7.6 Android

Not working:
- 7.7 Android
- 7.8 Android

Expected behavior
-----------------

1. Info -> Weather -> METAR and TAF -> Add
2. Type ICAO code -> OK
3. Select Airport -> Update
4. METAR and TAF data should show up

Actual behavior
---------------

It says 'No METAR available'. I tried multiple airports.

Steps to reproduce the behavior
-------------------------------

I tried on Android 7.6, 7.7, 7.8 versions, 7.6 is working perfectly. 

Every other download options are working on 7.8, e.g. I can download airspace and map files with the download button

On 7.8 I can see the following in the log while hit update in XCSoar:
```
06-03 13:23:08.826 I/XCSoar  (9741): CURL failed: error:0A000152:SSL routines::unsafe legacy renegotiation disabled
```

Do you have any idea what may have caused this?
-----------------------------------------------

My device is on Android 11, I see there are other issues with connection to internet with newer Android versions, maybe it's related?

Do you have an idea how to solve the issue?
-------------------------------------------

No

lordfolken, 2021-06-27:
curl -v -n https://tgftp.nws.noaa.gov/data/observations/metar/decoded/LSZH.TXT
```bash
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0x562fb1f59ee0)
* Connected to tgftp.nws.noaa.gov (140.172.138.79) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: OU=Domain Control Validated; CN=dev.ncep.noaa.gov
*  start date: Jan  7 16:31:34 2021 GMT
*  expire date: Jan 27 15:33:01 2022 GMT
*  subjectAltName: host "tgftp.nws.noaa.gov" matched cert's "tgftp.nws.noaa.gov"
*  issuer: C=US; ST=Arizona; L=Scottsdale; O=GoDaddy.com, Inc.; OU=http://certs.godaddy.com/repository/; CN=Go Daddy Secure Certificate Authority - G2
*  SSL certificate verify ok.
```

This seems fairly modern:
SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384

MaxKellermann, 2021-07-08:
No, the server is really broken. This went unnoticed because old OpenSSL versions had `SSL_OP_LEGACY_SERVER_CONNECT` enabled by default, but 3.0.0 disables it by default. Compare:
https://www.openssl.org/docs/man1.1.1/man3/SSL_clear_options.html
> The option SSL_OP_LEGACY_SERVER_CONNECT is currently set by default even though it has security implications: otherwise it would be impossible to connect to unpatched servers (i.e. all of them initially) and this is clearly not acceptable.

https://www.openssl.org/docs/manmaster/man3/SSL_clear_options.html
> Setting the option SSL_OP_LEGACY_SERVER_CONNECT has security implications; clients that are willing to connect to servers that do not implement RFC 5746 secure renegotiation are subject to attacks such as CVE-2009-3555. OpenSSL client applications wishing to ensure they can connect to unpatched servers should always set SSL_OP_LEGACY_SERVER_CONNECT

(The wording "set by default" has been removed in "master" / 3.0.0)

MaxKellermann, 2021-07-08:
https://github.com/openssl/openssl/commit/72d2670bd21becfa6a64bb03fa55ad82d6d0c0f3

closed 2021-07-08 by MaxKellermann