XCSoar/XCSoar #642: Assertion failed on AllocatedGrid.hxx:124

ubx, 2021-06-03:
XCSoar versions having and not having the problem
-------------------------------------------------

version: 7.9-Linux
git: 85b6494cc94abbe602cbb2be4fabf80f99771019

Expected behavior
-----------------

Render the map.

Actual behavior
---------------
Crash:

``
xcsoar: ./src/util/AllocatedGrid.hxx:124: AllocatedGrid<T>::const_iterator AllocatedGrid<T>::GetPointerAt(unsigned int, unsigned int) const [with T = TerrainHeight; AllocatedGrid<T>::const_iterator = const TerrainHeight*]: Assertion `x < width' failed.
``

Stack trace:
``
__GI_raise 0x00007fe046d6bfbb
__GI_abort 0x00007fe046d51864
__assert_fail_base 0x00007fe046d51749
__GI___assert_fail 0x00007fe046d633d6
AllocatedGrid<TerrainHeight>::GetPointerAt AllocatedGrid.hxx:124
RasterBuffer::GetDataAt RasterBuffer.hpp:65
RasterBuffer::Get RasterBuffer.hpp:83
RasterBuffer::ScanLine RasterBuffer.cpp:240
RasterTile::ScanLine RasterTile.hpp:139
RasterTileCache::ScanTileLine ScanLine.cpp:313
RasterTileCache::ScanLine ScanLine.cpp:345
RasterMap::ScanLine RasterMap.cpp:125
HeightMatrix::Fill HeightMatrix.cpp:92
RasterRenderer::ScanMap RasterRenderer.cpp:197
``

Values:
`width = {unsigned int} [256](url)
height = {unsigned int} 256
x = {unsigned int} 980
y = {unsigned int} 64967`

Steps to reproduce the behavior
-------------------------------

Compile XCSoar:
`make TARGET=UNIX OPENGL=n ENABLE_SDL=y USE_SDL2=y GEOTIFF=n DEBUG=y`

1. Start: `./output/UNIX/bin/xcsoar -portrait`
2. SIM mode
3. Select Profile default.prf
4. Config ... Replay
5. Select Replay File:  2021-03-07-XCS-AAA-01.igc
6. Start

Do you have any idea what may have caused this?
-----------------------------------------------
If I change Configuration -> Map Display -> Orientation -> Cruise orientation to **North up**  the problem goes away.
With **Track up** it crashes!

[default.prf](https://github.com/XCSoar/XCSoar/files/6593310/default.prf.TXT)

[2021-03-07-XCS-AAA-01.igc](https://github.com/XCSoar/XCSoar/files/6593314/2021-03-07-XCS-AAA-01.igc.TXT)

lordfolken, 2021-06-05:
From IRC:
<User48> Wanna submit bug on Win 64b version
<User48> versions 7.7, 7.6 and 7.5 crashing when setting/map display is set to 'Heading up'
<User48> 7.4 works great
<User48> Map display/orientation/Cruise orientation

ubx, 2021-06-18:
After some investigation and debugging, I came to this workaround:
```
diff --git a/src/Terrain/RasterBuffer.cpp b/src/Terrain/RasterBuffer.cpp
--- a/src/Terrain/RasterBuffer.cpp	(revision d329b936cc77e41c847e5acdff30df2a74852d0a)
+++ b/src/Terrain/RasterBuffer.cpp	(date 1624031306109)
@@ -217,8 +217,9 @@
      pixels in our buffer; the factor of two should account for the Y
      axis, which can have a different scale, making the factor some
      sort of ugly kludge to avoid horizontal shading stripes */
-  if (interpolate &&
-      (unsigned)(abs(d.x) + abs(d.y)) < (2 * size << RasterTraits::SUBPIXEL_BITS)) {
+//  if (interpolate &&
+//      (unsigned)(abs(d.x) + abs(d.y)) < (2 * size << RasterTraits::SUBPIXEL_BITS)) {
+  if (interpolate) {
     /* interpolate */
 
     for (int i = 0; (unsigned)i <= size; ++i) {
```
It's not a fix, it's just a hint!

ubx, 2021-06-21:
Better stack trace formatting.

MaxKellermann, 2021-06-29:
Fixed by ec2a0e08fb3d10096f84c53da8cb6bb8e7a0871f

closed 2021-06-29 by MaxKellermann