XCSoar/XCSoar #644: System crash on startup - Kobo Glo HD with Blueflyvario

stratocondra, 2021-06-05:
XCSoar versions having and not having the problem
7.6 and 6.8.16

Expected behavior
Normal start after receiving a GPS fix

Do you have any idea what may have caused this?
no

Do you have an idea how to solve the issue?
no

lordfolken, 2021-06-05:
Some more info maybe? Screen shots logfiles ,?

stratocondra, 2021-06-06:
Hi Philipp,

I haven't done any screenshots assuming there is no much to record. I had a
quick look at the crashes logfile, but couldn't see any relevant record. I
am not much of a software expert, so I am not sure if I was looking in the
right place. If you can guide me I can try to replicate the issue again and
gather more data.
Just to give you an idea about myself - I used to build these units few
years ago in quite a number so I am not a newbie to XCSoar, just more
involved on the hardware side.

Cheers,
Pav

lordfolken, 2021-06-06:
Logfile: the xcsoar data directory should have an xcsoar.log.

Is the vario soldered? Does it happen without the vario attached or configured in the profile.prf fike?

stratocondra, 2021-06-19:
> configured in the profile.prf file?

Hi Philipp,

Apologies for a long pause in communication.

It appears that I have found a reason for a crash. After changing terrain
map file, the system is stable. New file is an old backup of the
problematic map, so looks like the map file got corrupted in some way
during system update. It is however interesting that corrupted map file can
crash the system.

Cheer,
Pav

lordfolken, 2021-06-19:
That's interesting indeed. Could you provide ac copy of the file?

stratocondra, 2021-06-19:
Hi Philipp,
I thought you would ask this question. Unfortunately I deleted the
corrupted file and because I  did it straight off the device it is not even
in the recycling bin on my pc. Sorry,
Pav

lordfolken, 2021-06-19:
That's too bad. Improving the parser would have been interesting. Please reopen if needed.

closed 2021-06-19 by lordfolken