XCSoar/XCSoar #648: TCP client's do not reconnect

iltis42, 2021-06-18:
XCSoar version 6.8, mostly a generic issue in all other version, tested 6.3, 6.4, 7.0 several previews.
Connecting external device as TCP client, with external IP address via port 8880, 8881, etc.

Expected behavior
-----------------
After TCP connection e.g. via WLAN went out of range temporarily, XCSoar should automatically reconnect to the corresponding port. The corresponding port/device may go down and after stated "No Data", it should come into "Connected" mode again after the connection issue is fixed.

Actual behavior
---------------
At that time the device will stay permanently in state "No Data", and we need manually press "Reconnect" in the device manager for the corresponding port to bring up TCP connection again into state "Connected". Immediately after the we press this, connection comes up again and XCSoar shows "Connected: Baro, Vario, etc...". If there are multiple ports, we need to do this for each port.

Steps to reproduce the behavior
-------------------------------
Connect XCSoar using TCP client interface to any device supporting this, e.g. an XCVario is supporting this. A WLAN connection to produce temporary outage might used, or in case of a fixed TCP cable connection, remove cable and insert cable again, or reboot/restart the TCP source.

Do you have any idea what may have caused this?
-----------------------------------------------
Possibly keep alive supervision is missing or doesn't work. As "Reconnect" restores the connection, i guess this trigger is not operational, existing. No issue in Bluetooth, and no issue in LK8000 with common roots to reconnect a TCP client port. 

Do you have an idea how to solve the issue?
-------------------------------------------
Not yet, guess checking if there is a keep alive supervision for TCP clients, and what should be triggered in case of a connection failure.

lordfolken, 2021-06-18:
Could you please verify if this is still an issue in 7.8? Previews of 7 where very old.

iltis42, 2021-06-18:
The version 7.8 IMHO doesn't exist, if you mean 6.8, then clearly yes.

MaxKellermann, 2021-06-18:
Yes it does exist, see https://play.google.com/store/apps/details?id=org.xcsoar
This bug report is bogus, because it applies to **very** old versions which are unsupported.

iltis42, 2021-06-18:
Sorry, may fault, and thanks for mentioning! 
I meant 7.8 of course, but had written 6.8. Corrected this in new bugreport.
https://github.com/XCSoar/XCSoar/issues/650

hjr, 2021-06-20:
Yip, definitely present on 7.7, second this here!

closed 2021-06-20 by lordfolken

hjr, 2021-06-23:
Hey Philipp,

thought you solved this, but no, and why did you closed the issue again. It is really a fun breaker and even reproducible with head git rev on my laptop, when connecting as a TCP/IP client through Wifi.
disconnect/reconnect the Wifi connection and XCS will sit there not receiving any data until you press the "reconnect" button on the devices configuration page.
Must be something easy to fix.

thx,

Helmut