XCSoar/XCSoar #649: Android 9: power off screen on eInk devices (Kobo like)

hjr, 2021-06-18:
Hey guys,

XCSoar version
--------------
v7.7

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
I'd like to contribute with a Kobo like power off device screen for eReaders with eInk display. There will be more devices/screens that support a none powered screen display in the near future. The Kobo idea to display the last e.g. 10 take-off and landing times/dates is very handsome. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Did any of you have knowledge on where to start reading? What is the API on an Android device to set the power-off screen display?

Thx,

Helmut

lordfolken, 2021-06-18:
The function displaying the logs is here:
https://github.com/XCSoar/XCSoar/blob/464df291983948c67e967725e613b25c3efd9814/src/Kobo/PowerOff.cpp#L95

The kobo being linux based, there isn't really a high sophisticated process. Its simply the last screen being drawn and then the linux binary /sbin/poweroff is called. 

So a permission would have to be requested:
https://stackoverflow.com/questions/34102139/how-to-make-action-shutdown-intent-work-when-app-is-closed-in-android#34102226

But from there i'm at a loss. The Problem will be that it redraws the screen on shutdown which is probably an OS function...

hjr, 2021-06-20:
Hi Philipp,

thanks a lot. That was the pointer I was looking for. 

To dedicate generally an Android eReador for XC usage it would need on
top of this a specialized app to be started by default after boot (Kobo
like). Dont know if there is something like this.

Thx,

Helmut

lordfolken, 2021-06-20:
I think a launcher replacement works be a possibility.

lordfolken, 2021-07-02:
I'm closing this, as this is an idea.

closed 2021-07-02 by lordfolken