XCSoar/XCSoar #650: TCP client doesn't reconnect

iltis42, 2021-06-18:
XCSoar version 7.8, mostly a generic issue in all other version, tested 7.3, 7.4, also from 7.0 several previews had been tested.
Connecting external device as TCP client, with external IP address via port 8880, 8881, etc.

Expected behavior
After TCP connection e.g. via WLAN went out of range temporarily, XCSoar should automatically reconnect to the corresponding port. The corresponding port/device may go down and after stated "No Data", it should come into "Connected" mode again after the connection issue is fixed.

Actual behavior
At that time the device will stay permanently in state "No Data", and we need manually press "Reconnect" in the device manager for the corresponding port to bring up TCP connection again into state "Connected". Immediately after the we press this, connection comes up again and XCSoar shows "Connected: Baro, Vario, etc...". If there are multiple ports, we need to do this for each port.

Steps to reproduce the behavior
Connect XCSoar using TCP client interface to any device supporting this, e.g. an XCVario is supporting this. A WLAN connection to produce temporary outage might used, or in case of a fixed TCP cable connection, remove cable and insert cable again, or reboot/restart the TCP source.

Do you have any idea what may have caused this?
Possibly keep alive supervision is missing or doesn't work. As "Reconnect" restores the connection, i guess this trigger is not operational, existing. No issue in Bluetooth, and no issue in LK8000 with common roots to reconnect a TCP client port.

Do you have an idea how to solve the issue?
Not yet, guess checking if there is a keep alive supervision for TCP clients, and what should be triggered in case of a connection failure.

henrik-the, 2021-06-22:
I am experiencing the same issue.

hjr, 2021-06-23:
It is really a fun breaker and even reproducible with head git rev on my laptop when connecting as a TCP/IP client through Wifi.
Disconnect/reconnect the Wifi connection and XCS will sit there not receiving any data until you press the "reconnect" button on the devices configuration page.
Must be something easy to fix.

thx

lordfolken, 2021-06-24:
Possibly https://github.com/XCSoar/XCSoar/commit/f295a3cca4f07b4683c8040351f636d95eaa36f5 ?

iltis42, 2021-06-25:
Maybe the asio trouble. From News.txt, i see long ago: Version 6.7.2 - 2013/12/19
  - auto-reconnect TCP client (#3127)
this had been trouble already ín the past. The history and also the old issue number is no more in this repo, maybe sm1 has this still locally to look what had been done at that time?

closed 2021-06-29 by MaxKellermann

iltis42, 2021-06-29:
Superb, great new's Max!

hjr, 2021-06-30:
Wow, could confirm with my laptop that the TCP auto reconnect is perfectly fixed.

lots of kudos Max!