XCSoar/XCSoar #651: Background logging not working on Android 10

mkauzlar, 2021-06-24:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem? 
7.8
Which XCSoar version is the last one you know that did not show the problem? 
not sure but it was working in previous versions of Android ( 9)

Expected behavior
-----------------

Please describe what you expected to see/should happen.
Run XCSoar and close the screen and leave it running through the flight. It should not stop.

Actual behavior
---------------

Once started it's on for some time and then, when the screen is put on again in the drop-down notifications it says that XCSouar is running, however clicking there it brings the XCSoar initial page where it asks for FLY or SIM.
This happens every time I try to use XCSoar as a logger (running in the pocket). If the display is on then it runs all the time.

Steps to reproduce the behavior
-------------------------------

1. Install XCSOar
2. allow location all the time
3. disable battery optimisation
4. run in fly mode
5. record movement for some time and close the display. 
6. Wait some time (I'm not sure how long before it stops), meaning if I do it within 1 minute it still works.
7. reopen the display and click on "XCSoar is running" notification tab
8. XCSoar is back to the initial screen without loggings 

Do you have any idea what may have caused this?
-----------------------------------------------

Battery optimization and other settings that might stop it in the background are off

Do you have an idea how to solve the issue?
-------------------------------------------

No

MaxKellermann, 2021-06-24:
Did you allow XCSoar to access your GPS location while running in background?

mkauzlar, 2021-06-24:
Yes, as I mentioned in step 2, I selected "allow location all the time".
The other options are "allow only when using the app" and "deny"

MaxKellermann, 2021-06-24:
What does "back to the initial screen without loggings" mean? Are you back at the "Fly/Sim" selection screen?
If yes, then this means XCSoar has crashed. In that case, extract a crash log ("logcat"). Without the crash log, there is nothing I can do to fix this.

mkauzlar, 2021-06-25:
@MaxKellermann yes that is correct it goes back to "FLY/SIM", sorry for my poor wording.

This test was done in a room without GPS signal available. I don't know if that matters.
Here is the logcat output. XCSoar started around 8:19 and crashed around 8:26
[xcsoar-logcat.zip](https://github.com/XCSoar/XCSoar/files/6714052/xcsoar-logcat.zip)

MaxKellermann, 2021-06-25:
No, it does not crash. It is forcibly killed by a Blackview-specific process ("Bob.chen") which appears to kill background apps:
```
06-25 08:26:08.292  1111 29883 E Bob.chen: ClearMemorySyncTask. doInBackground. begin kill recentlist. numTasks =1
06-25 08:26:08.292  1111 29883 D Bob.chen: ClearMemorySyncTask. doInBackground. recent[0], id = 2706, persistentId = 2706, packageName = org.xcsoar
06-25 08:26:08.293  1111 29883 D Bob.chen: ClearMemorySyncTask. doInBackground. intent.getComponent().getPackageName() = org.xcsoar. exist_in_whitelist = 0
06-25 08:26:08.293  1111 29883 E Bob.chen: RecentTasksLoader. ClearMemorySyncTask. begin kill: recentInfo.persistentId = 2706
06-25 08:26:08.298 30281 30281 D XCSoar  : in onDestroy()
```
Not a XCSoar problem.

closed 2021-06-25 by MaxKellermann

mkauzlar, 2021-06-25:
ok, thank you for looking into this

mkauzlar, 2021-06-28:
If somebody ends up here, the issue is with the Blackview phones (Android 10 and 11).

There is a system app called System Manager whic kills apps in the background after 5min.

In order to let XCSoar run in the background and possibly other apps (such as clock widgets etc) you need to:

Open System Manager app -> App Management -> Auto Cleanup and then select (enable) XCsoar

lordfolken, 2021-06-28:
Thx for reporting back. Potentially we have to ask for disabling power optimization in our manifest.

lordfolken, 2021-06-28:
https://stackoverflow.com/questions/39256501/check-if-battery-optimization-is-enabled-or-not-for-an-app#54982071