XCSoar/XCSoar #652: LeGPSBip+ connection

OttoPh, 2021-06-24:
Welcome! - We kindly ask that you:

  1. Fill out the issue template below - not doing so needs a good reason.
  2. Use the forum if you have a question rather than a bug or feature request.

The forum is at: https://forum.xcsoar.org

Thanks for understanding, and for contributing to the project!

-->

XCSoar version
--------------

 The latest 

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

The new LeGPSBip+ seems a great device to use in connection, but it seems XCSoar won't connect to that. If that is so, can it be added? Thanks! 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

 Connect a decent vario with GPS and loud sounds

MaxKellermann, 2021-06-24:
What is it?

OttoPh, 2021-06-24:
An external vario. 

See https://www.lebipbip.com/buy/index.php?id_product=26&controller=product&id_lang=13

⁣
Groet,
Otto

​

MaxKellermann, 2021-06-24:
I don't see any technical documentation which would allow us to support the product. If the manufacturer were interested in XCSoar compatibility, he could send us code to support it. Or at least give us one exemplar for free (if there is a volunteer developer willing to write code in his free time to support this commercial product).

But since there isn't even documentation, there's nothing we can do. Ask the vendor for support.

closed 2021-06-24 by MaxKellermann

schmidi, 2021-06-24:
@OttoPh Have a look at this link: https://www.lebipbip.com/lk8000-for-android/ in the screenshot there is also a connectivity option for xcsoar. I guess there is no further implementation in xcsoar necessary...

MaxKellermann, 2021-06-24:
One more reason to demand XCSoar support from the hardware manufacturer, and not LeGPSBip+ support from XCSoar!

OttoPh, 2021-06-25:
Danke Tobias,

Ich befürchte, wenn ich das lese, das es sich hier nur um LK8000 
handelt.
Danke aber für das mitdenken!
Ich habe den LeBip bestellt und werde es ausprobieren.

Gruß,

Otto Philipse

------ Origineel bericht ------

Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

dave-prosee, 2021-06-25:
Hi,

Same request here but with a different angle; IMHO not the device specific content but physical layer is the issue.

I have an Android 10 device supporting OTG (can be checked with an app) and LeGPSBip works on it with LK8000 and XCtrack. But with XCSoar I am missing the USB OTG option to make it work (only IP, bluetooth and IOIO is offered).
I have seen work done for KOBO devices using USB OTG so could this USB OTG port option become available? The LeGPSBip emulation modes will do the rest.

elgandoz, 2021-06-28:
In order to use leGPSBip+ with XCSoar you must select the protocol in the GPSBip configurator (as displayed in the LK8000 pic). 
In XCSoar it must be configured as it was a BlueFlyVario with the USB connection.
Check this page, under the section XCSoar:
https://www.lebipbip.com/xcsoar-for-kobo/#1463929657804-69b38c4a-20b4

MaxKellermann, 2021-06-30:
XCSoar 7.9 will have support for USB serial adapters, like the one leGPSBip+ uses.

dave-prosee, 2021-07-01:
> XCSoar 7.9 will have support for USB serial adapters, like the one leGPSBip+ uses.

Thanks, I am sure it will do the trick ;-)

MaxKellermann, 2021-07-05:
7.9 is on Google Play. Try it.
(Or here: https://download.xcsoar.org/releases/7.9/ANDROID/XCSoar.apk)

dave-prosee, 2021-07-07:
Dear Max,
Thanks for putting OTG in XCsoar 7.9. It is working on my android10 device with LeGPSBip. The port noticed the device and it was selectable. Baudrate had to be set at 4800. And for driver I chose BlueFly Vario (as LeGPSBip also mentioned for use on Kobo device with XCSoar). So your work paid off. Vario is in the display. Only need to check if the gps is working, but I see no reason why it should not. You made my day.

MaxKellermann, 2021-07-07:
Yay!