XCSoar/XCSoar #655: XCSoar autostart by IOIO-connection, the touchbuttons do not function

abiweil, 2021-06-29:
XCSoar versions having and not having the problem
-------------------------------------------------

I'm using XCSoar version 7.7, I did not try it with other versions.
The used device is a ONYX poke2 color eInk Reader with Android 9 on board.

Expected behavior
-----------------

After switching on the device with a connected IOIO-device (Soartronic), normally a window with the heading-line XCsoar and the following question appears: "for IOIO XCSoar standard application open?" After pushing the "OK"-button XCSoar should be opened.

Actual behavior
---------------

After pushing the "OK"-button the normal XCSoar-screen appears as it should be, but the touchbuttons "FLY" or "SIM" on the base XCSoar screen show no reaction. The only button which is active, is the "Quit"-button.
After pushing the "Quit"-button, appears a windows with the heading "IOIO standard application" and the message: No installed apps for this USB-accessory. Further informations under https://github.com/ytai/ioio/wiki/ADK.
If you finish the window by pushing "Escape", you can launch the XCSoar-app with the complete functionality included the IOIO-functions.

If you push the "Escape"-button instead of the "OK"-button, you can launch the XCSoar-app manually, the "FLY" or "SIM"-buttons in this case are reacting, but after that the above mentionned window with the headline XCSoar appears again. Now you can push the "OK"-button and XCSoar shows its normal functionality.

Steps to reproduce the behavior
-------------------------------
By restarting the device, the behaviour is always the same.

If you plug-in the USB-connector in an already running device, the mentionned window with the XCsoar headline appears as usual and by pushing the "OK"-button all things function normally.

Do you have any idea what may have caused this?
-----------------------------------------------
perhaps a timing-problem between activating the touchfunctions and the USB-connection?

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2021-06-30:
I'm unable to reproduce this problem. The Fly/Sim buttons always work, even if XCSoar is auto-launched by plugging into a IOIO.

abiweil, 2021-06-30:
Hallo, Max,
It seems to be a special problem with the ONYX poke2 color. If I switch over to the homepage of the poke2 (independent from the way to do it) and then back to XCSoar, which is still running in the background, all things work fine. 
The homepage of the poke2 color is a fixed homepage (the ebook-library), which cannot be changed in the system-stettings of the poke2 color. In the black and white-version, you have obviously the possibility to change the homepage.
If the poke2 is already booted I can switch-on the IOIO-interface and XCSoar is launched without any problems.
It seems to be a timing problem. My theory is, that XCSoar is intitiated, before the Android system is completely booted. But I’m not a software expert, only electronic engineer!
By the way, the poke2 color is an ideal device for XCSoar which has been recommended by Helmut Rohs.
With regards or mit Fliegergruß
Albert Kiessling

From: Max Kellermann 
Sent: Wednesday, June 30, 2021 5:03 PM
To: XCSoar/XCSoar 
Cc: abiweil ; Author 
Subject: Re: [XCSoar/XCSoar] XCSoar autostart by IOIO-connection, the touchbuttons do not function (#655)

I'm unable to reproduce this problem. The Fly/Sim buttons always work, even if XCSoar is auto-launched by plugging into a IOIO.

MaxKellermann, 2021-08-13:
Since this appears to be a vendor-specific bug and not a XCSoar bug, and no developer has such a device, it will be impossible to find a workaround. Sorry, this isn't going to be fixed.

closed 2021-08-13 by MaxKellermann