XCSoar/XCSoar #656: Compiling with Docker is failing due to missing sources.

ulipo, 2021-07-02:
XCSoar versions having and not having the problem
-------------------------------------------------
V7.8, GIT Master branch...

Expected behavior
-----------------

Docker environment should let compile XCSoar out of the box

Actual behavior
---------------

Compiling is failing due to [ide/provisioning/install-debian-packages.sh](https://github.com/XCSoar/XCSoar/blob/master/ide/provisioning/install-debian-packages.sh) not installing some dependencies needed to compile GCC, specifically GMP, MPC and MPFR sources

Steps to reproduce the behavior
-------------------------------

- Clone GIT repo
- start docker compiling script for KOBO
- wait for it...

Do you have any idea what may have caused this?
-----------------------------------------------

Missing GMP, MPC and MPFR sources in output/src/gcc-11.1.0/ 

Do you have an idea how to solve the issue?
-------------------------------------------

A workaround is to manually download the sources and place them in place, following directions at https://gcc.gnu.org/install/prerequisites.html

MaxKellermann, 2021-07-02:
Can you submit a PR which patches the script to install libgmp-dev etc.?

ulipo, 2021-07-02:
I'm sorry but I don't know how to patche the script, I just downloaded the sources by hand and put them in place...

MaxKellermann, 2021-07-02:
Then you've done something much more complicated - it's much easier than that. All you need is add a line `apt install libgmp-dev ....` (figure out which -dev packages you need on Debian) and an explanatory comment.

closed 2021-07-02 by lordfolken