XCSoar/XCSoar #663: xcsoar entire screen is red/pink

michielb88, 2021-07-10:
XCSoar versions having and not having the problem
-------------------------------------------------

7.10-android on android 6. Hw: phone CAT S60
On the previous version ( 6.xx ) i did not had any issue

Actual behavior
---------------

The entire screen have a kind of red/pink wash over it. Also over the menu. Only when turning the device there is a very short moment the screen colour is with normal colours

Steps to reproduce the behavior
-------------------------------

This red/pink wash is always applicable 

Do you have any idea what may have caused this?
-----------------------------------------------

No idea

Do you have an idea how to solve the issue?
-------------------------------------------

No idea

michielb88, 2021-07-10:
[image]
[image]
While making a screenshot suddenly the normal colours appear. 
So I also added an oldskool screenshot were my issue is visible.

brunotl, 2021-07-13:
if that can help, i have fixed the same probleme in LK8000 few years ago, see  commit https://github.com/LK8000/LK8000/commit/e2fb79dd

MaxKellermann, 2021-07-21:
Thanks @brunotl, do you have a device that reproduces the problem? To me, it seems like a driver bug, but I'd like to know before applying such a workaround which affects all devices, even those which are not buggy. It would be interesting to know which EGL configuration is chosen on these buggy devices.

brunotl, 2021-07-22:
unfortuantly i don't have device to reproduce anymore. 

my patch is just a partial revert of change you have made in thsi commit https://github.com/XCSoar/XCSoar/commit/d6ebfe40
i don't know why you have changed the resquest from 565 to 444...

about choosing EGL Config, maybe you can also take a look to this commit : https://github.com/LK8000/LK8000/commit/ac55e6155c77bf764f61c0982329fa5b2a3849b9

MaxKellermann, 2021-07-22:
That's indeed a bad commit (by me) - badly documented and mixes two changes.
I think I thought I'd allow more different color depth configurations - I vaguely remember there was one problem which got fixed by that.
Reverting this (like you did) would fix one, but break the other. Not a satisfying solution to anything. I want to know the real problem. If it's really a bad GPU, we need to explicitly blacklist the broken modes, but only for that GPU, to avoid breaking others. Sigh.

MaxKellermann, 2021-07-22:
Indeed my commit was bad - I confused "minimum RGB depth" and "desired RGB depth". But anyway, this leading to a rendering problem still looks like a GPU (driver) bug, and I hope we have a workaround which works everywhere. With the revert by @brunotl, the code is certainly better.

closed 2021-07-22 by MaxKellermann

michielb88, 2021-07-22:
Thanks a lot for fixing! Any idea in which release it will be included and do you have a timeline?

MaxKellermann, 2021-07-22:
@brunotl your other commit https://github.com/LK8000/LK8000/commit/ac55e6155c77bf764f61c0982329fa5b2a3849b9 looks rather useless, or even harmful. Giving `EGL_STENCIL_SIZE=1` means this is the minimum number of stencil bits; EGL will not return any configuration with less than that.
The old code prefers less stencil bits (i.e. less memory waste), but your code prefers many stencil bits (i.e. more memory waste). In practice, this most likely doesn't make a difference because the stencil depth is usually always 0 or 8 bits, but in theory, your change is bad. If ever any GPU appears which supports, say, 16 bits of stencil, your code will choose 16 instead of 8.

brunotl, 2021-07-22:
> @brunotl your other commit [LK8000/LK8000@ac55e61](https://github.com/LK8000/LK8000/commit/ac55e6155c77bf764f61c0982329fa5b2a3849b9) looks rather useless, or even harmful. Giving `EGL_STENCIL_SIZE=1` means this is the minimum number of stencil bits; EGL will not return any configuration with less than that.
> The old code prefers less stencil bits (i.e. less memory waste), but your code prefers many stencil bits (i.e. more memory waste). In practice, this most likely doesn't make a difference because the stencil depth is usually always 0 or 8 bits, but in theory, your change is bad. If ever any GPU appears which supports, say, 16 bits of stencil, your code will choose 16 instead of 8.

Thanks @MaxKellermann, you right, could be better to revert it...