XCSoar/XCSoar #664: FlarmNet file not working

Turbo87, 2021-07-13:
XCSoar versions having and not having the problem
-------------------------------------------------

Android, latest release. Last time I used the v6 I think it was still working, but not sure if it's related to the v7 update.

Expected behavior
-----------------

Competition IDs should be read from the data.fln file and shown on map and flarm radar

Actual behavior
---------------

They don't seem to show up anymore

Steps to reproduce the behavior
-------------------------------

Download the latest data.fln file and put it in the XCSoarData folder. Start XCSoar, go to the Traffic List in the Info menu. It should be possible to search for known competition IDs there.

Do you have any idea what may have caused this?
-----------------------------------------------

Unfortunately I don't

Do you have an idea how to solve the issue?
-------------------------------------------

Haven't found anything useful in the logs :(

lordfolken, 2021-07-13:
I'm able to search for comp IDs in that dialog, without issue. I tried both flarm and ogn. 

Lacking currently a flarm I cannot check the map part. However the pilot and frequency fields remain suspiciously empty.

Turbo87, 2021-07-13:
Hmm strange. Did you download a new file? I tried manual download and via file manager and both did not seem to work. Friend of mine has the same issue on his android phone 🤔

lordfolken, 2021-07-14:
[image]

lordfolken, 2021-07-17:
@Turbo87 do you have the same behavior?

Turbo87, 2021-07-17:
I wasn't aware that 7.x made you explicitly select the file. I was expecting it to automatically be picked up as the default, like it was with 6.x.

lordfolken, 2021-07-18:
Yes, like this its at least consistent with all the other file selections.

Great, i just discovered that we where serving the ogn db as flarmnet since ... 2018. No one complained and I'm to blame.
https://github.com/XCSoar/xcsoar-data-repository/commit/070fbe7923f84f664b2a01a81ddab3b021c6ae0c

closed 2021-07-18 by lordfolken

Turbo87, 2021-07-18:
Yeah, it's fine to have the field like that, but it would have been nice to select the file by default if the key does not exist in the profile 😉

lordfolken, 2021-07-18:
The whole file selection process is a mess. Some files get discovered, some don't. Its very inconsistent.

lordfolken, 2021-07-18:
You would need some logic to discover both files. You would need to handle the case where both files are there.

Turbo87, 2021-07-18:
> You would need to handle the case where both files are there.

not sure if that's true. the 6.x logic was "if `data.fln` exists, use it". it didn't care about the ogn file at all AFAIK. so it would make sense to use `data.fln` as the default, if it exists.