XCSoar/XCSoar #668: GoTo should display 2 segments to waypoint

scls19fr, 2021-07-18:
XCSoar version
--------------

XCSoar 7.10

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

When using GoTo waypoint function, a segment (with arrow) between current position and waypoint is drawn.

It could be nice if 2 segments to waypoint could be drawn (with differents colors or grayscale).

We could see for example :
- a black segment (with arrow) between current position and designed waypoint (as currently) 
- an other segment (light grey) from the position where the GoTo function have been activated to the designed waypoint.

Such a drawing could greatly help to see if our trajectory deviates from straight line (without creating a task).

MaxKellermann, 2021-07-20:
What would be the point of that? Returning to the original (grey) line would only add more detour.

scls19fr, 2021-07-20:
It won't be a HSI (horizontal situation indicator) but it will help to see if glider deviates much from grey line or not

MaxKellermann, 2021-07-20:
That doesn't answer my question. What would you gain from knowing that your glider deviated from the original grey line? It's not anymore the shortest route at this point. The shortest route is always the black line. The grey line shows obsolete information.

scls19fr, 2021-07-20:
It doesn't help for future decisions but it enlight past as it shows that I could (theorically) have done better.
The display of trajectory is also obsolete information. Obsolete doesn't mean useless.

csindle, 2021-08-23:
Knowing one's crosstrack error (XTE, or XTK) can be very useful to fly the shortest distance to a waypoint when the wind is unknown/changing, and XCSoar cannot give the correct heading to fly. This is a simple method to avoid the classic navigation problem of constantly heading towards one's waypoint, yet ending up approaching it from due downwind (and flying further/longer).

References:

 ArduPilot's main HUD has another way to display this crosstrack info: item 2:
https://ardupilot.org/planner/docs/mission-planner-ground-control-station.html

https://www.roboticboat.uk/navigation.html

Search for XTE: https://www.yachtingmagazine.com/thinking-your-gps-system/