XCSoar/XCSoar #674: Android split screen support

r0w4n, 2021-07-28:
XCSoar version
--------------

7.10

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Android supports split screen, using two apps at the same time. Currently xcsoar will crash when going into spilt screen mode 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Using xcsoar and other apps simultaneously e.g artificial horizon.

elgandoz, 2021-07-31:
It works fine here on XCSoar 7.11 and Samsung S10e on Android 11.
[image]

MaxKellermann, 2021-07-31:
If XCSoar ever crashes, then that is the real problem and should be reported. XCSoar must never ever crash.

r0w4n, 2021-07-31:
Hello,

I'm using android 11 / Oxygen OS 11.0.1.1.HD01BA

I've screen recorded the issue. Please see it on YouTube 

https://youtu.be/6JjTRBteJKI

MaxKellermann, 2021-08-05:
Your video shows that XCSoar was running successfully in split screen.

r0w4n, 2021-08-05:
The video shows xcsoar crashing several times before it finally runs

MaxKellermann, 2021-08-05:
I don't see any crash in your video. If there is a crash, prove it by posting the crash log.

r0w4n, 2021-08-05:
I've made another screen cast to show the crash, hopefully clearer this
time. The first attempt to run xcsoar along side chrome results in xcsoar
terminating. The second attempt xcsoar starts up and doesn't terminate.

https://youtu.be/J1yvyCM3F8U

MaxKellermann, 2021-08-06:
There is no crash in your video and no crash in the huge file you sent me.

r0w4n, 2021-08-06:
I'm kind of confused as to why you're not seeing the crash in the video. In
the video I

1) start chrome
2) Initiate split screen
3) start xcsoar
4) xcsoar terminates almost immediately
5) start xcsoar for a second time
6) this time xcsoar runs.

MaxKellermann, 2021-08-06:
XCSoar terminates itself from the startup screen as soon as you switch to another app. That's not a crash, that's desired behavior to avoid cluttering your phone's resources. You havn't even started XCSoar at this point. As soon as you actually start XCSoar, it does not terminate itself anymore until you explicitly quit it.

r0w4n, 2021-08-06:
The starting condition is that xcsoar is not running. After starting chrome
and starting the split screen mode I start xcsoar for the first time. It
quits immediately. I then start it for the second time, at which point it
runs.

MaxKellermann, 2021-08-06:
> After starting chrome and starting the split screen mode I start xcsoar for the first time

This is a completely different thing than your initial bug report:

> Currently xcsoar will crash when going into spilt screen mode

None of your videos show what this bug report is about - XCSoar never crashed when going into split screen mode. You caused this confusion because your bug report was not accurate.

r0w4n, 2021-08-06:
Until you explained that xcsoar was in fact closing itself I was under the
impression that it was crashing. If you replace crash with close
unexpectedly then I believe that my report is accurate.

This closing behaviour is not desirable from my perspective.

MaxKellermann, 2021-08-06:
Not desirable, but your report was not accurate. You were not switching to split screen and that caused XCSoar to terminate itself; you first switched to split screen, then started XCSoar, and then XCSoar terminated itself. These details do matter, that's not splitting hairs, that's utterly important.

r0w4n, 2021-08-06:
Would you like me to write this up as a bug report, accurately describing
the steps?

MaxKellermann, 2021-08-06:
I understood now, I just wanted to point out that I'd like you to consider those tiny details when writing a bug report next time.

closed 2021-08-06 by MaxKellermann