XCSoar/XCSoar #676: USB-OTG setting on Clara HD crashes and bricks the Kobo

Boulderpilot1, 2021-07-31:
XCSoar versions having and not having the problem
-------------------------------------------------

 7.10, I don't know if other versions work 

Expected behavior 
-----------------

 Device should be able to run data from XC tracer III GPS to Kobo Clara HD using USB-OTG connection, as described on XC Tracer website

Actual behavior
---------------

As soon as you select USB-OTG, a prompt occurs "USB-OTG not supported for Clara HD, use at own risk?" Select yes and then Kobo immediately crashes and bricks. I re-imaged the SD to original image and retried from the beginning. The same problem occurred twice. Kobo remains bricked.

Steps to reproduce the behavior
-------------------------------

 After new install of XCSoar on Clara HD, select system, then enable USB-OTG, when prompted select "Yes" - this leads to a crash/brick

Do you have any idea what may have caused this?
-----------------------------------------------

No ideas

Do you have an idea how to solve the issue?
-------------------------------------------

No ideas

MaxKellermann, 2021-08-23:
This is not a bug. The message clearly states that the Kobo Clara HD is not supported, and you installed an incompatible kernel despite the warning. Of course it does not work.
To support the Kobo Clara HD, we need a volunteer to develop this, and that volunteer needs to have a Kobo Clara HD.

sepi, 2021-09-01:
@Boulderpilot1 is the device recoverable? Usually the term "bricked" means that the device is broken beyond repair and is now as good as a brick.

GeekGoneOld, 2021-12-30:
This has been fixed in 7.21.  The Enable USB-OTG button is greyed out for Clara HD.  Check commit ae4a195 on src/Kobo/SystemDialog.cpp

@MaxKellermann or @lordfolken  perhaps can close this issue?

closed 2022-01-06 by GeekGoneOld