XCSoar/XCSoar #677: Incorrect Date and Time in version 7.11

ZbebV, 2021-08-02:
XCSoar versions having and not having the problem
-------------------------------------------------

I encountered the problem using XCSOAR 7.11.
I did not encounter the problem using XCSOAR 7.8

Expected behavior
-----------------

 I was expecting to have a flight log with the correct date in the name of the file and in the "HFDTE" field of the .igc file.
I was expecting the logged time in the IGC file to be correct. 

Actual behavior
---------------

In the IGC file name : 
wrong date : 01-01-1970 and 02-01-1970 instead of 01-08-2021 and 02-08-2021
in the igc file : 
wrong date in the "HFDTE" field : 010170 and 020170 instead of 010821 and 020821
wrong logged time for each logged position by around 5 hours.
After 2 flights with this issue, I looked in the date/hour parameters. UTC offset was OK (UTC +2) but the displayed time was still off by at least 5 hours. 

Steps to reproduce the behavior
-------------------------------

 Two consequent flights had the same symptoms so no specific action required to reproduce the problem.
I use the XCSOAR app on a Samsung A3 2017.

Do you have any idea what may have caused this?
-----------------------------------------------

I tried changing the time parameter to use the smartphones built in watch/clock in stead of the GPS time. But it did not match the phone clock even after restarting the app. 

Do you have an idea how to solve the issue?
-------------------------------------------

 No idea except the obvious issue with the GPS decyphering the GPS time and using the built in clock of the phone.
[1970-01-02-XCS-AAA-01.zip](https://github.com/XCSoar/XCSoar/files/6919412/1970-01-02-XCS-AAA-01.zip)

sigged, 2021-08-09:
Experiencing the same problem on a Samsung device. Other devices running version 7.8 are not affected.

oscar-goldman, 2021-08-09:
I also have the same problem with version 7.11. Year in IGC file is always 1970 and month is January (all the flights tested were in August 2021). My device is a Samsung Galaxy J3.

lordfolken, 2021-08-09:
hmm maybe: https://github.com/XCSoar/XCSoar/commit/ac0f570583a76e003a51bf869c57e4bae7de3e7c ?

closed 2021-08-10 by MaxKellermann