XCSoar/XCSoar #681: Flarm radar infobox not updating correctly

Turbo87, 2021-08-08:
XCSoar versions having and not having the problem
-------------------------------------------------

> Which XCSoar version are you using when seeing the problem?

v7.11

> Which XCSoar version is the last one you know that did not show the problem?

I'm not sure. v6 definitely did not have this issue. I've been using v7.10 before and I don't think I noticed it on that version either, but I could be wrong.

Expected behavior
-----------------

The Flarm radar infobox should update at least once per second, similar to the Flarm targets on the map.

Actual behavior
---------------

[image]

The Flarm radar infobox sometimes doesn't update for 10+ seconds, while the targets on the map are moving correctly.

Steps to reproduce the behavior
-------------------------------

Not sure if there is anything special about my setup. Connect a Flarm and fly with it, I guess... 😅 

Do you have any idea what may have caused this?
-----------------------------------------------

It seems that the incoming Flarm traffic messages are not causing the infobox to invalidate and rerender. Why it sometimes updates correctly and other times it doesn't, I don't know.

Do you have an idea how to solve the issue?
-------------------------------------------

Unfortunately, I don't.

MaxKellermann, 2021-08-12:
I can't reproduce this (with NMEA replay).
"Connect a Flarm and fly with it" is not an option for me.
If you can reproduce it, please do a bisect.

alextheinsomniac, 2021-08-24:
Hi Max,
I logged in to report a similar bug. I have recently been updated to 7.15 from last year's version 6.

I have been flying a competition with the screen highly zoomed in to be able to see flarm traffic because the dedicated flarm page says "No traffic" even though there is moving Flarm traffic on the map on the other pages. The flarm radar info box also doesn't seem to update. (Actually I would suggest that perhaps they get data every now and again, perhaps every couple of minutes.)

 I have enclosed a short video showing the issue which I recorded when other's were launching.

Whilst I probably can't gather any useful data this week, I should be able to sit near our launch point one day and capture something useful. Could you please let me know:
 a) What logs and other info would be of most help to you and
b) how I can test the files to ensure the issue is reproduced before I upload them here.

Thanks and Thanks for a great piece of software.

https://user-images.githubusercontent.com/4974229/130682291-84847053-d45a-46b9-8b78-2afb7a4d345b.mov

MaxKellermann, 2021-08-24:
@alextheinsomniac I need a NMEA dump of a flight that reproduces the problem. And try all XCSoar releases, to find out which was the very first which had the bug, and which one was the last which did not.

Scumi, 2021-08-25:
For me, the issue is not reproducible with replaying my NMEA files with 7.15 where I know the problem occurred. Tried on android and linux. 
Must have something to do with the live environment.

Turbo87, 2021-09-24:
I don't know why you assigned me. As @Scumi already reported, it seems that the issues unfortunately does not reproduce when replaying an NMEA log... 😞

MaxKellermann, 2021-09-24:
So you're unwilling to help debug your issue. What do you suggest should be done now, and who should do it?

Turbo87, 2021-09-24:
I am not unwilling, I am just clueless how to do it and my C++ have significantly degraded so I can't really do much

MaxKellermann, 2021-09-24:
So your problem turns into a gordian knot. I have clue but can't fly, and have no clue and you don't want to obtain clue. Means you have to wait until somebody happens to come around with the right combination of skills. Until then, this will remain `wontfix`.

lordfolken, 2022-05-20:
Please see #925 and related pulls. Therein its also described how to do a flarm replay.

closed 2022-05-22 by lordfolken