XCSoar/XCSoar #683: Time showing incorrectly in Fly mode

RossDouglasMc, 2021-08-13:
XCSoar versions having and not having the problem
7.11 has the problem.
-------------------------------------------------

Expected behavior    Show the correct time (GMT and Local
-----------------

Actual behavior
In Fly mode it is showing a GMT and Local time 4 hours and 42 minutes ahead of the actual time.
Works perfectly in Simulate mode.
---------------

Steps to reproduce the behavior
Tred resets and checked all settings. Does it every time I start in Fly mode. Does not do it in Simulate mode. 
-------------------------------

Do you have any idea what may have caused this?
None. It has just showed up today, which was the first time I have used it flying for some months. I have been using it in Condor over the winter.
-----------------------------------------------

Do you have an idea how to solve the issue?
Sorry, no.
-------------------------------------------

MaxKellermann, 2021-08-13:
What is the most recent version which does not have this problem? (Take the time and try them all.)
What device and operating system? Which GPS or other connected devices/sensors?

RossDouglasMc, 2021-08-13:
Hi, I am using a Samsung S5 phone with Android version 6.0.1. XC Soar is using the phone GPS. There are no other connected devices or sensors. XC Soar updates to new versions automatically and I don’t even notice that when it happens. My last soaring flight using XC Soar was on 18/04/2021. The system and time was working correctly then. I don’t know what version it was, but would have been the latest at that time. As I said, it is now working correctly in Simulate mode, but not in Fly mode. All other features appear to me working correctly in Fly mode, it is only the time that is incorrect. I don’t know how to revert to previous versions. RegardsRoss McLernon   From: Max KellermannSent: Friday, 13 August 2021 8:11 PMTo: XCSoar/XCSoarCc: RossDouglasMc; AuthorSubject: Re: [XCSoar/XCSoar] Time showing incorrectly in Fly mode (#683) What is the most recent version which does not have this problem? (Take the time and try them all.)What device and operating system? Which GPS or other connected devices/sensors?

MaxKellermann, 2021-08-13:
Download and install APK files from http://download.xcsoar.org/releases/ and try which was the first bad one.

RossDouglasMc, 2021-08-13:
Hi, Thanks.  I loaded 7.10 and then 7.9 and both worked as they should. Both showed the correct time, GMT and Local in Fly mode. Downloaded and reloaded 7.11 and it was still the same, showing the wrong time in Fly mode. So last version that worked was 7.10 Best RegardsRoss From: Max KellermannSent: Friday, 13 August 2021 9:48 PMTo: XCSoar/XCSoarCc: RossDouglasMc; AuthorSubject: Re: [XCSoar/XCSoar] Time showing incorrectly in Fly mode (#683) Download and install APK files from http://download.xcsoar.org/releases/ and try which was the first bad one.

MaxKellermann, 2021-08-13:
Duplicate of https://github.com/XCSoar/XCSoar/issues/677

closed 2021-08-13 by MaxKellermann

RossDouglasMc, 2021-08-14:
Thanks very much. I see this morning version 7:12 with the time issue fixed. Once again, absolutely outstanding service from you. Greatly appreciated. Best RegardsRoss McLernon From: Max KellermannSent: Friday, 13 August 2021 10:26 PMTo: XCSoar/XCSoarCc: RossDouglasMc; AuthorSubject: Re: [XCSoar/XCSoar] Time showing incorrectly in Fly mode (#683) Closed #683.