XCSoar/XCSoar #684: Invalid UTC time report in Info->Status->Times window when date&time source is SensBox BLE

lyusupov, 2021-08-14:
XCSoar versions having and not having the problem
-------------------------------------------------

7.12 Android ; target CPU is ARM64

Expected behavior
-----------------

UTC date and time fields should report current UTC D&T.

Actual behavior
---------------

[image]

Steps to reproduce the behavior
-------------------------------

Connect the XCSoar 7.12 to SensBox over BLE, wait for a GNSS fix,
then proceed into Info->Status->Times window.

Do you have any idea what may have caused this?
-----------------------------------------------

**When 'Built-in GPS & sensors' source is selected - the date and time values are fine**.

Not sure, but the reason for that behavior could be an improper 32->64 bit time conversion while executing 
listener.onLocationSensor():
https://github.com/XCSoar/XCSoar/blob/master/android/src/BluetoothSensor.java#L122

According to the commits log, there were certain date & time logic improvements been applied recently.
This could cause a regression of the particular BLE sensor code.

Do you have an idea how to solve the issue?
-------------------------------------------

No.

MaxKellermann, 2021-08-14:
This is the buggy code line:
https://github.com/XCSoar/XCSoar/blob/9b34760abfd763d8d2d9fe148dc70407cd11e62c/android/src/BluetoothSensor.java#L122
This passes a UNIX time stamp in seconds to a function which expects a UNIX time stamp in milliseconds.
Care to write a PR?
When you do, be sure to cast to `long` first before multiplying with 1000.
And for bonus points, take care for making it (somewhat) Y2038 safe ;-)
What a mess, somebody designs a device in the 21st century with only 32 bits for a UNIX time stamp... I wonder if it wraps over or just crashes in 2038.

lyusupov, 2021-08-14:
> Care to write a PR?

Unfortunately, I am unable to do that - I don't have the Android build environment readily available here. :-(