XCSoar/XCSoar #688: Not possible to declare task to LXNAV S100

zweefan, 2021-08-16:
XCSoar versions having and not having the problem
-------------------------------------------------

XCSOAR version 7.11 on android
S100 version 8.02 (is not working)
nano 3.00 (is working)
nano3 3.02 (is working)

Expected behavior
-----------------

task -> declare should result in declaring the task into the S100.
All other communication with the S100 is working (connection over bluetooth)
Similary, the task declarition to an lxnav nano and nano3 is working.
I think download of igc files is also not working with the s100 (other but similar issue)

Actual behavior
---------------

task is not declared, after a while an error is displayed "task not declared"

Steps to reproduce the behavior
-------------------------------

connect with BT to lxnav S100
in task->manage press declare task

Do you have any idea what may have caused this?
-----------------------------------------------

much be some (recent?) change in the LXNAV protocol.
previously (i heard from somebody) it used to be working.

Do you have an idea how to solve the issue?
-------------------------------------------

contact lxnav for latest protocol

MaxKellermann, 2021-08-16:
No XCSoar developer has a S100 for driver development. This cannot be solved until one is made available. You could contact LXNAV and ask them to provide one for the project.

zweefan, 2021-08-16:
I have one and i can test.
Asking lxnav if there has been a protocol change might be the first step.
Creating a hw simulator in software the next one.
Testing against the real physical device after that.
Just a suggestion...

MaxKellermann, 2021-08-16:
You can test, that's good (that way, you can contribute to XCSoar by testing each new release with it, allowing developers to spend their time on other things), but for proper hardware support, a XCSoar driver developer needs one such device.
A simulator in software would be great, but who will spend time developing one?

alextheinsomniac, 2021-08-24:
Zweefan . I can declare to a S100 (and in my case on to a connected flarm mouse) provided I follow a simple procedure. I've been using the S100 since they first shipped (~5 years ago) and this always works for me.

You need to go config/devices/reconnect then immediately close the dialogue and go nav/task/manage/declare.  I'm not sure how long you have to do it but provided I don't stop it always works for me. No need to rush. If I don't reconnect just before then I seem to remember it fails by hanging for ages and then giving an error.

Please let us know how you get on.

Alex

zweefan, 2021-08-30:
I can confirm that the workarround provided by Alex is indeed working.
I tried several times and declaring immediatly after a "reconnect" works.
The task is transmitted. The turnpoint area's are however not transmitted but i guess that an official declaration only includes the waypoint cooridnates (and not the obeservation zones).
Instead the S100 uses the default zone's as configured in the device.

Wouldn't it be possibe to force a reconnect whenever a declaration is made to a LXNAV device?
I have multiple LX nav devices available (nano, nano3, S100 and potentially a lx-9000) so i can test on all these devices.

MaxKellermann, 2021-08-30:
> Wouldn't it be possibe to force a reconnect whenever a declaration is made to a LXNAV device?

Everything is possible, but a developer would first need to find out what is the best solution for this problem.

PR #728 (2021-09-28): Fix for LXNAV Varios
Brief summary of the changes
----------------------------
Fixes issues with declaring tasks and other comms with LXNAV S series varios.

Based on original work from LXNAV located here: https://github.com/XCSoar/XCSoar/pull/220/

Tested with an LXNAV S100 vario. No testing has yet been performed with other LXNAV Varios or their passthrough capability with the Nano.

Related issues and discussions
------------------------------
Closes #688 

Update to indicate test steps taken with S100 vario:

1. NAV->Task->Manage->Declare; XCSoar indicates success, S100 receives task.
2. C [... cut]
closed 2021-09-28 by MaxKellermann