XCSoar/XCSoar #689: Suggestion for a new vario Gauge

Vantuilmv, 2021-08-16:
XCSoar version

7.11

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

What are you trying to do, what is the use case for the suggestion?

New vario gauge

Suggestion for a new vario meter, it can use 2 infobox in the same way as the current one, and have a configuration option to choose the type of meter, follow a model I made, I think it would be very interesting. 
[image]

MaxKellermann, 2021-08-16:
How does this new design work, and why do you believe it is better than the old one?
I don't know what your graphic means.

Vantuilmv, 2021-08-16:
I use XCSoar on kobo, this model that I proposed would be a way to visualize it more easily, in principle all rectangles would be unfilled, each rectangle would represent 1 m/s, in the center we would have the value of the snapshot (Gross ), in the upper part the average of the thermal and in the lower part the MC or other information that I find interesting, for example the gain of the thermal, I will make some drawings with values to explain it better. But I don't think about removing the other meter, just adding an alternative. 

[image]

[image]

closed 2021-11-18 by Vantuilmv