XCSoar/XCSoar #690: android navigation bar/buttons stay on top

zweefan, 2021-08-18:
XCSoar versions having and not having the problem
-------------------------------------------------
problem in 7.12
in 7.11 it was still ok

Expected behavior
-----------------
when switching from another application to xcsoar (already running in the background),
The android navigation bar should be hidden.

Actual behavior
---------------
The navigation bar stays on top and it is actually impossible to press the xcsoar buttons beneath it.

Steps to reproduce the behavior
-------------------------------
start xcsoar
switch to some other application
switch back to xcsoar

Do you have any idea what may have caused this?
-----------------------------------------------
no

Do you have an idea how to solve the issue?
-------------------------------------------

some code in 7.12 broke this. in 7.11 and before it wass still ok

MaxKellermann, 2021-08-18:
You didn't post any information about your device. Which one, which Android version?

fronzee, 2021-08-18:
I also just tried on Samsung galaxy s9. Problem exists.

zweefan, 2021-08-18:
tested on two devices
aquila, android 6
oneplus 6T, Android 10

zweefan, 2021-08-19:
[image]

closed 2021-08-19 by MaxKellermann