XCSoar/XCSoar #694: XCSoar v7.13 would not start on Android11/Pixel4a

MikeBravo15, 2021-08-19:


MikeBravo15, 2021-08-19:
XCSoar v7.13 get stuck at initial screen - not buttons (Fly/Sim/Quit) are functional

MaxKellermann, 2021-08-19:
Can't reproduce with any of my Android devices.

schmidi, 2021-08-19:
I have the same problem here on my Pixel 4a 5G with Android 11. See video...

https://user-images.githubusercontent.com/411354/130116041-e7d85fdb-80a2-4721-b23f-0149b8f20a4b.mp4

MaxKellermann, 2021-08-19:
Does clicking slightly above those buttons work?

schmidi, 2021-08-19:
> Does clicking slightly above those buttons work?

Yes, a bit above works!

https://user-images.githubusercontent.com/411354/130119419-ffa5ff72-5d78-4383-bab3-1081fa76c607.mp4

https://user-images.githubusercontent.com/411354/130119455-7e26ada2-9457-44ef-b701-9cbc2819b9e1.mp4

Scumi, 2021-08-19:
For me the buttons work weirdly if I set the fullscreen option to "OFF". Also, XCSoar is still in fullscreen, but the OS buttons overlay the XCSoar screen (instead of reducing the size of XCSoar).

MaxKellermann, 2021-08-19:
I have found a Galaxy S8 in my drawer which has similar problems.

closed 2021-08-20 by MaxKellermann

schmidi, 2021-08-20:
@MaxKellermann Thanks for the fix!