XCSoar/XCSoar #695: XCSoar v7.13 crashes on startup while loading terrain on Android 6/Aquila

bhudson225, 2021-08-19:
Device: Aquila running Android 6.0
Installed from Google Play Store
Crash prevents startup when terrain starts to load, closing app. Multiple pilot profiles tried, cache and app data cleared, app reinstalled from fresh. 7.12 and earlier versions work.

MaxKellermann, 2021-08-19:
Need a crash dump (logcat).

lyusupov, 2021-08-20:
I have the same issue with **Android 5.1** (on ARM64).
XCSoar 7.12 was just fine.  But both **7.13** and **7.14** do no longer boot up.

This is an excerpt from 'logcat':

```

I/XCSoar  (32310): Startup dialog
D/SQLiteDatabase(29519): beginTransaction()
D/SQLiteDatabase(29519): endTransaction()
I/XCSoar  (32310): Loading profiles
D/SQLiteDatabase(29519): beginTransaction()
I/XCSoar  (32310): Loaded profile from /storage/emulated/0/XCSoarData/default.prf
D/SQLiteDatabase(29519): endTransaction()
D/Surface (32310): Surface::setBuffersDimensions(this=0x7fa564e600,w=1080,h=1845)
I/BufferQueueProducer(  272): [SurfaceView](this:0x7f8ab27000,id:662,api:1,p:32310,c:272) new GraphicBuffer needed
I/[MALI][Gralloc](  272): Usage: 0xb00, format: 0x4 stride: 1088 vertical_stride: 1845 size: 4014720
D/GraphicBuffer(  272): alloc, handle(0x7f84532ca0) (w:1080 h:1845 s:1088 f:0x4 u:0x000b00) err(0)
D/GraphicBuffer(32310): register, handle(0x7f8d416020) (w:1080 h:1845 s:1088 f:0x4 u:0x000b00)
D/SQLiteDatabase(29519): beginTransaction()
I/BufferQueueProducer(  272): [SurfaceView](this:0x7f8ab27000,id:662,api:1,p:32310,c:272) queueBuffer: fps=1.85 dur=1081.57 max=967.61 min=113.96
D/PowerManagerService(  883): acquireWakeLockInternal: lock=98504288, flags=0x1000001a, tag="XCSoar", ws=null, uid=10175, pid=32310
I/thermal_repeater(  321): oh, queryMdThermalInfo (0)Bad file number
D/SQLiteDatabase(29519): endTransaction()
I/thermal_repeater(  321): [recvMdThermalInfo] ret=26, strLen=127,  1, 32767,,, -1, 32767, 18
D/PowerManagerService(  883):   |----com.android.server.power.PowerManagerService.wakeUpNoUpdateLocked(PowerManagerService.java:1480)
D/PowerManagerService(  883):   |----com.android.server.power.PowerManagerService.applyWakeLockFlagsOnAcquireLocked(PowerManagerService.java:1079)
D/PowerManagerService(  883):   |----com.android.server.power.PowerManagerService.acquireWakeLockInternal(PowerManagerService.java:1049)
D/PowerManagerService(  883):   |----com.android.server.power.PowerManagerService.access$4900(PowerManagerService.java:105)
D/PowerManagerService(  883):   |----com.android.server.power.PowerManagerService$BinderService.acquireWakeLock(PowerManagerService.java:3587)
D/PowerManagerService(  883):   |----android.os.IPowerManager$Stub.onTransact(IPowerManager.java:66)
D/PowerManagerService(  883):   |----android.os.Binder.execTransact(Binder.java:451)
D/PowerManagerService(  883): wakeUpNoUpdateLocked: eventTime=35183327, uid=10175
D/PowerManagerService(  883): updateWakeLockSummaryLocked: mWakefulness=Awake, mWakeLockSummary=0x2b
D/PowerManagerService(  883): updateUserActivitySummaryLocked: mWakefulness=Awake, mUserActivitySummary=0x1, nextTimeout=35236183 (in 52822 ms)
D/PowerManagerService(  883): mDisplayPowerRequest.policy = 3
D/PowerManagerService(  883): isWfdEnabled() mWfdStatus=WifiDisplayStatus{featureState=2, scanState=0, activeDisplayState=0, activeDisplay=null, displays=[[TV][LG]27MT75S-PZQ (ca:02:10:5f:8c:45), isAvailable false, canConnect false, isRemembered true], sessionInfo=WifiDisplaySessionInfo: C/O:Client,GId:,Pp:,SId:0,IP:}, return false
D/DisplayPowerController(  883): requestPowerState: policy=BRIGHT, useProximitySensor=false, screenBrightness=71, screenAutoBrightnessAdjustment=0.39600003, useAutoBrightness=true, blockScreenOn=false, lowPowerMode=false, boostScreenBrightness=false, dozeScreenBrightness=-1, useSettingBrightness=false, dozeScreenState=UNKNOWN, waitForNegativeProximity=false
I/PowerManagerService(  883): setBrightness mButtonLight, screenBrightness=71
D/PowerManagerService(  883): updateDisplayPowerStateLocked: mDisplayReady=true, policy=3, mWakefulness=1, mWakeLockSummary=0x2b, mUserActivitySummary=0x1, mBootCompleted=true, mScreenBrightnessBoostInProgress=false
D/PowerManagerNotifier(  883): onWakeLockAcquired: flags=268435482, tag="XCSoar", packageName=org.xcsoar, ownerUid=10175, ownerPid=32310, workSource=null
D/AndroidRuntime(32310): Shutting down VM
E/AndroidRuntime(32310): FATAL EXCEPTION: main
E/AndroidRuntime(32310): Process: org.xcsoar, PID: 32310
E/AndroidRuntime(32310): java.lang.NoSuchMethodError: No virtual method isInMultiWindowMode()Z in class Lorg/xcsoar/XCSoar; or its super classes (declaration of 'org.xcsoar.XCSoar' appears in /data/app/org.xcsoar-1/base.apk)
E/AndroidRuntime(32310):        at org.xcsoar.XCSoar.wantFullScreen(XCSoar.java:187)
E/AndroidRuntime(32310):        at org.xcsoar.XCSoar.applyFullScreen(XCSoar.java:192)
E/AndroidRuntime(32310):        at org.xcsoar.XCSoar$4.handleMessage(XCSoar.java:182)
E/AndroidRuntime(32310):        at android.os.Handler.dispatchMessage(Handler.java:111)
E/AndroidRuntime(32310):        at android.os.Looper.loop(Looper.java:194)
E/AndroidRuntime(32310):        at android.app.ActivityThread.main(ActivityThread.java:5691)
E/AndroidRuntime(32310):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(32310):        at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime(32310):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:959)
E/AndroidRuntime(32310):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:754)
V/SettingsInterface(  883):  from settings cache , name = dropbox:data_app_crash , value = null
V/SettingsInterface(  883):  from settings cache , name = send_action_app_error , value = 1
D/ActivityManager(  883): SVC-mBroadcastQueues: com.android.server.am.BroadcastQueue@1c21b083
[... cut ...]
W/ActivityManager(  883):   Force finishing activity 1 org.xcsoar/.XCSoar
V/WindowManager(  883): Changing focus from Window{33c2f930 u0 org.xcsoar/org.xcsoar.XCSoar} to null Callers=com.android.server.wm.WindowManagerService.setFocusedApp:4703 com.android.server.am.ActivityManagerService.setFocusedActivityLocked:2591 com.android.server.am.ActivityStack.adjustFocusedActivityLocked:2932 com.android.server.am.ActivityStack.finishActivityLocked:3189
I/WindowManager(  883): Losing focus: Window{33c2f930 u0 org.xcsoar/org.xcsoar.XCSoar}
```

MaxKellermann, 2021-08-20:
OK, `Activity.isInMultiWindowMode()` https://developer.android.com/reference/android/app/Activity#isInMultiWindowMode() was added in API level 24 (Android 7), and XCSoar should check the Android version before calling it.
Sadly this comes just after I released 7.14 - and I don't have time for another release this weekend. So all users with Android before 7.0 need to wait.

closed 2021-08-23 by MaxKellermann