XCSoar/XCSoar #696: Extend OpenVario Device Driver for Wind and AHRS

MaxBaex, 2021-08-20:
XCSoar version    7.09

Extend the OpenVario Device Driver  for 
- Wind
- AHRS information

Existing Interface:
   * E: TE vario in m/s
   * H: relative humidity in %
   * P: static pressure in hPa
   * Q: dynamic pressure in Pa
   * R: total pressure in hPa
   * S: true airspeed in km/h
   * T: temperature in deg C
   * ?: respond with selected strings, e.g. Ballast, Bugs, Polar
   *    Example: $POV,?,RPO,MC,WL*2E means: Real Polar, MacCready, Wing Load

Added Interface:
  * W:  Windspeed in m/s
  * D:   True North Winddirection in °   
  * I:    Pitch Angle    in °    Front / Right / Down
  * J:    Roll Angle 
  * K:   True North Heading Angle
  * 

It might be usefull to extend the interface so that two or more values can be provides after a single letter. E.g. Windspeed and Direction after W.  Pitch, Roll, Heading after A. 

We want to connect an experimental variometer via Bluetooth and aditionally provide wind and AHRS information. This should all be integrated in one driver. 

Do you think extending this driver is a reasonable way to providing Wind and AHRS data to XCSOAR from one single driver / device?

MaxKellermann, 2021-08-23:
I don't understand this request. Do you want XCSoar to support an already existing new revision of the protocol published by the OpenVario project, or do you want to extend the OpenVario protocol?

MaxBaex, 2021-08-23:
I want to extend the existing OpenVario protocol. I am not aware of a new protocol revision by the OpenVario project.

MaxKellermann, 2021-08-23:
Then better talk to the OpenVario project first. This is the wrong place to add proprietary extensions to the protocol; it should first be documented by the OpenVario project, and then implemented in XCSoar.

closed 2021-08-23 by MaxKellermann

MaxBaex, 2021-08-23:
Thanks for the quick reply! 

I evaluated the different device driver protocols. I rather reuse/extend an existing one then creating another device driver. The protocol lacks wind and AHRS support. Wind information is supported independently of the configured driver via $GPMWV strings. I am not sure if a device driver should additionally provide wind information?  

Is there an intention to create a standard variometer device driver?  We are not bound to the OpenVario device driver but this one seems to be a good starting point.