XCSoar/XCSoar #700: Livetrack24 forwarding

Tontilano, 2021-08-28:
XCSoar versions having and not having the problem
-------------------------------------------------

I have seen the problem from 1th of August on. Last working version was 7.10 on 28th of July. Later on I tried an other App with the same settings and it worked. Please see https://www.livetrack24.com/user/flytietar01 and https://www.dhv-xc.de/leonardo/index.php?op=list_flights&pilotID=0_239&takeoffID=0&country=0&year=0&month=0&season=0

Expected behavior
-----------------

forwarding flight to livetrack24

Actual behavior
---------------

Data transfer to Skylines live but not to Livetrack24

Steps to reproduce the behavior
-------------------------------

There is no way to make XCSoar to make it forwarding data to livetrack24. Meanwhile it was not possible to avoid XCSoar to fall in sleep mode, but this is solved now.

Do you have any idea what may have caused this?
-----------------------------------------------

No Idea

Do you have an idea how to solve the issue?
-------------------------------------------

Tontilano, 2021-08-28:
XCSoar does not forwarding my tracks to livetrack24 anymore.

MaxKellermann, 2021-08-29:
I just tried with version 7.10 and it does not work. My suspicion is that LiveTrack24 changed their protocol this month.
Please downgrade to 7.10 and try again, and confirm that it still works with that version!

MaxKellermann, 2021-08-29:
You can find all old releases here: https://download.xcsoar.org/releases/

Tontilano, 2021-08-29:
7.10 cannot be installed. Phone denies. A friend of mine uses still 6.8.14 and it works perfectly.
https://www.livetrack24.com/user/FlyTietar02

MaxKellermann, 2021-08-30:
> I just tried with version 7.10 and it does not work

I tried again with 6.8.14, and it did not work as well. But I found the problem: I have been testing with server `test.livetrack24.com`, and that server is obviously broken, just look at the front page: https://test.livetrack24.com/

After switching to the main server `www.livetrack24.com`, tracking did work perfectly, see https://www.livetrack24.com/track/1981223

So... I cannot reproduce your problem.

Tontilano, 2021-08-30:
I never set "test.livetrack24.com".

Today all day long I installed and uninstalled 6.8.14, 7.10 ans 7.16. I also changed settings. The result was always the same Skylines server received tracking data, Livetrack24 did not.

Meanwhile i tried TTLivetrack24 app - livetrack24 combination with the same settings and it worked.  No idea where the tracking data are getting lost.

MaxKellermann, 2021-08-30:
> Today all day long I installed and uninstalled 6.8.14, 7.10 ans 7.16. I also changed settings. The result was always the same Skylines server received tracking data, Livetrack24 did not.

This contradicts with the information you gave in your initial bug report: you said 7.10 works. And later you said 6.8.14 works for your friend. Now you say neither 6.8.14 nor 7.10 nor 7.16 works. That's confusing.

Tontilano, 2021-08-30:
I said 7.10 was the last working version in my case (28.07.2021). After my tests today I realized, now, none of the today tested versions work anymore. Something has happened between 28.07. and 01.08.. I suspected a fault in location sharing permission, what I check precisely since I realized, it wasn't possible to set for sleeping modes for a while for XC-Soar.
My friend installed 6.8.14 and never updated. It has worked without problems. Last trial on 28.08. and I'm sure it will work tomorrow as well. (off course, his Skyline transmission does not work (this is an other report)).

MaxKellermann, 2021-08-30:
Not a XCSoar regression in any case, if 7.10 did work but does not now. This bug report is bogus.

closed 2021-08-30 by MaxKellermann