XCSoar/XCSoar #701: Forwarding track to DHV-XC-(Leonardo)-server

Tontilano, 2021-08-28:
XCSoar version
--------------

7.16

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Forwarding track to DHV-XC-(Leonardo)-server

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

At the end I want to score via DHV-XC. With this feature it would be possible to avoid errors on passing the tracks on via livetrack24.

MaxKellermann, 2021-08-29:
I need technical information on how to do this - which servers will be used, and which protocols are available. And I need an account for testing the code.

Tontilano, 2021-08-29:
This is the server url:
 https://www.dhv-xc.de/
 This is more information about:
 https://www.dhv-xc.de/leonardo/index.php?op=program_info
 My account:
 https://www.dhv-xc.de/leonardo/index.php?op=pilot_profile&pilotIDview=0_239
 access data via mail.
 I'm only user. I don't have programming skills

MaxKellermann, 2021-08-30:
None of these links contain any technical information about how XCSoar is supposed to transmit data to this server.

https://www.dhv-xc.de/leonardo/index.php?op=program_info links to http://www.leonardoxc.net/ but that's just an empty page saying:
> ad block detected
> Please disable your ad blocker and reload the page.

WTF, an open source project makes money with ads (after copying code from XCSoar), and rejects people with ad blockers - which strange parallel universe is that?
That is annoying.

MaxKellermann, 2021-08-30:
btw. XCSoar does already have an option to submit to `livexc.dhv.de` instead of `www.livetrack24.com` (you can choose in the XCSoar configuration), but that server is offline. There is nothing XCSoar can do to bring this server back online.

Tontilano, 2021-08-30:
livexc.dhv.de isn't supported anymore and died. I refer to forward the track to dhv-xc.de after flight is closed, for scoring, like livetrack24 does if it is set in configuration.

MaxKellermann, 2021-08-30:
> I refer to forward the track to dhv-xc.de after flight is closed

On `dhv-xc.de`, live tracks cannot be submitted. Again: I need reliable technical information, and the information you just told me is wrong.

Tontilano, 2021-08-30:
You are right. livlracks can't. It would only work if the track would be stored in a device or a server and would be submited to DHV-XC after flight is closed. Livetrack24 server does this, but has problems in recognizing of flight end (see https://www.dhv-xc.de/leonardo/index.php?op=compare&flightID=1430614,1430587)
However, it would be a nice gimmic, but nothing important.

MaxKellermann, 2021-08-30:
I still need technical information on how to submit tracks.

Tontilano, 2021-08-30:
[image]
Lifetrack24 does it this way. When Lifetrack24 detects the end of a flight, it generates a igc file and loads it up the way you see in the screen sot. My skills in managing this things are very poor. I'm not sure wheter I can give reliable information. But at the moment thanks for your patience.

MaxKellermann, 2021-08-30:
This is not how LiveTrack24 does it. This is LiveTrack24's web interface.
Don't post random irrelevant stuff here, because non-technical discussion just wastes your time and my time, but won't bring this forward. Get me technical information, nothing less.

MaxKellermann, 2021-08-31:
Let's reopen this as soon as there is enough information.

closed 2021-08-31 by MaxKellermann