XCSoar/XCSoar #702: Topology layer affects airspace visibility

frankvdh, 2021-08-29:
Problem present in XCSoar versions 7.6 and 7.16, Linux & Android
I'm not sure when this started happening, but suspect some time before 7.6. I think it was OK at v6.8. If there's an archive of older releases, I'm happy to try to identify the start.

Expected behavior
-----------------
Airspace is displayed at all zoom levels, not affected by topology configuration

Actual behavior
---------------
Airspace disappears when zooming in or out. 

This is somehow affected by my Heliport and Airstrip topology layers. If I comment both of them out in topology.tpl, airspace displays as expected at all zoom levels. Also, moving the Heliport & Airstrip lines to the beginning of the topology.tpl file fixes the problem. There is nothing remarkable about these Shapefiles... they're both valid and list a number of points. I have other similar Shapefiles (e.g. SpotHeight, Mast) with more points.

If it is caused by the Airstrip layer, airspace is visible at 54nm zoom level, but zooming in it disappears at 40nm zoom level until it reappears at the 5.4nm zoom level. If it is Heliport causing the fault, these zoom levels are 81nm and 8.1nm. Screenshots below:... airspace is coloured cyan.

[image] [image]
[image] [image]

Steps to reproduce the behavior
-------------------------------
1. Configure XCSoar to use the attached VFR.xcm (renamed to .zip to be able to attach). The only airspace in this particular file is one CFZ just north of Palmerston North -- I deleted the rest because I thought it might be due to some error in the airspace.txt file. When I had multiple airspace layers, all were affected.
2. Set Home waypoint to Palmerston North via Nav/Waypoint List/Search... screen 3/Set as New Home
3. Zoom in and out... 

topology.tpl file that causes the fault, Heliport and Airstrip error-causing lines near the bottom:
```
* filename, range, icon, label_index, r, g, b, pen_width, label_range, label_important_range, alpha
water_area,100,,1,85,160,255,1,100
water_line,50,,1,85,160,255,1,50
roadbig_line,15,,1,192,48,48,5,5,
roadmedium_line,6,,,192,24,24,2,,
roadsmall_line,3,,,192,24,24,1,,
railway_line,10,,,64,64,64,1,10,0
Mast,10,obstacle,1,0,0,0,1,10,5
SpotHeight,8,mountain_top,1,0,0,0,1,10
Nativebig,50,,,63,127,63,,,,92
Exoticbig,50,,,31,127,31,,,,92
Nativemed,25,,,63,127,63,,,,92
Exoticmed,25,,,31,127,31,,,,92
Scrub,50,,,127,127,63,,,,92
ScatScrub,25,,,127,127,63,,,,92
Orchard,25,,,48,127,64,,,,92
Vineyard,25,,,64,127,48,,,,92
City,100,,1,255,127,127,1,50,20,192
Town,50,,1,255,127,127,1,25,15,192
Village,25,,1,255,127,127,1,25,10
GolfCourse,8,,,127,255,127,,,
Mainland,500,,,63,63,127,3,,
Islands,50,,1,63,63,127,3,10,4
RaceCourse,8,,,127,127,63,2,,
Power,25,,,0,0,0,2,,
Heliport,15,mountain_pass,1,0,0,0,1,15
runways,3,,,0,32,0,,,,64
PLA,5,target,1,0,0,0,1,3,1
RHCircuits,3,,,192,96,0,5,128
Airstrip,10,outfield_reachable,1,0,0,0,1,8,1

```
Do you have any idea what may have caused this?
-----------------------------------------------
I suspect some sort of leakage of the visibility limits from the topology rendering to the airspace rendering. There's nothing in the developer manual to say that the topology.tpl file format has changed.

Do you have an idea how to solve the issue?
-------------------------------------------
I expect code inspection/analysis and a bug fix???

For anyone else who encounters this issue there is a workaround of rearranging lines within topology.tpl.

MaxKellermann, 2021-08-29:
Where is this attached VFR.xcm?

MaxKellermann, 2021-08-29:
https://download.xcsoar.org/releases/

frankvdh, 2021-08-29:
Hi Max,
I'm not sure why the attachment failed... when I try to attach it as a VFR.xcm.zip or VFR.zip or VFR.ZIP or vfr.zip, I get a red "is not included in the list" at the bottom of the text panel. Here's a dropbox link to the file.
https://www.dropbox.com/s/2zxgik0bxekqpqq/VFR.xcm?dl=0

I've checked, and my file works properly with v6.8.17, but not with v7.5. There's no Linux version of v7.4, so I haven't tested that. My assumption is that it was the change from v6.8 to v7 that introduced the bug.

Frank

MaxKellermann, 2021-08-30:
Bug confirmed with your map file, very strange bug indeed.

closed 2021-08-30 by MaxKellermann