XCSoar/XCSoar #704: F-Droid build failed

linsui, 2021-08-31:
```
download https://download.libsodium.org/libsodium/releases/libsodium-1.0.18.tar.gz
download https://www.openssl.org/source/openssl-3.0.0-beta2.tar.gz
download https://c-ares.haxx.se/download/c-ares-1.17.1.tar.gz
download https://curl.se/download/curl-7.78.0.tar.xz
download http://www.lua.org/ftp/lua-5.4.3.tar.gz
download http://download.osgeo.org/proj/proj-5.1.0.tar.gz
download http://download.osgeo.org/libtiff/tiff-4.0.10.tar.gz
download http://download.osgeo.org/geotiff/libgeotiff/libgeotiff-1.4.2.tar.gz
touch output/ANDROID/lib/stamp
rm output/ANDROID/opt/src/Device/Driver/../dirstamp
```

https://monitor.f-droid.org/builds/log/org.xcsoar/137

Could you please take a look? Thanks!

MaxKellermann, 2021-08-31:
This one:
```
/home/vagrant/build/srclib/cpython/Modules/_ctypes/_ctypes.c:107:17: fatal error: ffi.h: No such file or directory
```
... no idea, this seems to be because F-Droid's build scripts do strange things to set up Python. Not related to XCSoar.
But this one:
```
g++: error: unrecognized command line option ‘-std=c++20’; did you mean ‘-std=c++03’?
```
XCSoar has recently switched to C++20, which requires GCC 10 or clang 10. The Android NDK comes with clang12, but one program needs to be compiled for running on the build host, and apparently your build host has only GCC 6 (released 5 years ago). That's way too old!

MaxKellermann, 2021-08-31:
Since this can't be solved by the XCSoar project, I close this bug report.

closed 2021-08-31 by MaxKellermann

lordfolken, 2021-08-31:
Hmm this log is weird. It seems to me its trying to build the toolchain and xcsoar at the same time...

linsui, 2021-08-31:
Debian stretch support up to c++17 via clang 7. c++20 is too new. I don't want to build gcc from source so we'll wait until we switch to bulleye.

MaxKellermann, 2021-08-31:
Debian Stretch ... oldoldstable, blast from the past!

linsui, 2021-08-31:
It's still oldstable last week...