XCSoar/XCSoar #706: XCSoar cannot connect to IOIO via ADB

bhudson225, 2021-09-03:
XCSoar versions having and not having the problem
-------------------------------------------------
Current (7.16) and several previous versions (all that were tested)

Expected behavior
-----------------
When connecting an IOIO board via USB-OTG, IOIO firmware and libraries compiled into XCSoar should communicate via Android Debug Bridge (if enabled) or OpenAccessory (if supported). UART data should be received by XCSoar regardless of connection method

Actual behavior
---------------
When an Android device is connected to an IOIO with debugging enabled, the "debugging connected" notification appears confirming that the IOIO is working in ADB mode. However, connections to the IOIO UART ports and battery sensor in XCSoar cannot be established - listed as "Not connected"

This is most relevant to the Aquila, which due to issues in the installed version of Android, cannot use OpenAccessory. However this incorrect behaviour has been observed additionally on an Android 8 device which can successfully connect via OpenAccessory.

Steps to reproduce the behavior
-------------------------------
Turn debugging on, connect to IOIO board, attempt to add IOIO UART as XCSoar device

Do you have any idea what may have caused this?
-----------------------------------------------
Potentially outdated or modified aspects of IOIOLib fork? Reading the documentation, IOIOLib should return data when requested for a given UART port regardless of whether ADB or OpenAccessory is used via its handshake process (which seems to work as far as establishing an ADB link between the devices)

Do you have an idea how to solve the issue?
-------------------------------------------
I am attempting to work out a way of testing the UART link to see if data can pass on ADB at all

MaxKellermann, 2021-09-03:
Which versions did you test? Please go back into the version history until you find a version which works.

MaxKellermann, 2021-09-06:
Also which IOIO firmware version?

bhudson225, 2021-09-06:
IOIO firmware is version 6.1.0, only recently been superseded by 6.2.0. The IOIO is currently in the panel - I will pull it out at the end of the month and do extended troubleshooting with various XCSoar and IOIO firmware versions. So far I've only tried with XCSoar 7.x versions, and not earlier.

bhudson225, 2021-09-25:
Have finished testing with major XCSoar versions back to 6.5 (as far as the .apk history goes). Have found:

- Can connect to the IOIO and receive data via OpenAccessory from 6.7 onwards
- Cannot receive data using ADB for any tested version (6.5.x - 7.1.x)

Can test earlier versions if able to access .apk's.

bhudson225, 2021-10-09:
Although not mentioned in the current documentation, ADB to IOIO has not been possible on Android versions higher than 4.2.2 as per https://groups.google.com/g/ioio-users/c/5gwna9Y6tGU. Therefore recommend closing the issue.

closed 2021-10-09 by bhudson225