XCSoar/XCSoar #707: After changing layout, layout is not updated when still in menu Config / System / Look

scls19fr, 2021-09-04:
XCSoar versions having and not having the problem
-------------------------------------------------

XCSoar 7.15 Android

Expected behavior
-----------------

After changing layout in menu Config / System / Look / Screen Layout
and going to InfoBoxes Sets / Circling it should show correct layout

Actual behavior
---------------

After changing layout, layout is not updated when we are in menu Config / System / Look

(ie when you don't go back to map)

Steps to reproduce the behavior
-------------------------------

DOWN-UP gesture 
Config / System / Look
Screen Layout
change InfoBox geometry

close

InfoBoxes Sets / Circling still displays previous layout

Do you have any idea what may have caused this?
-----------------------------------------------

Updating layout should occurs when closing "Screen Layout" page

Do you have an idea how to solve the issue?
-------------------------------------------

PS : changing language should also be updated before going back to map... (just when closing Look / Language, Input)

lordfolken, 2021-11-09:
Can you add some screenshots?

scls19fr, 2021-11-09:
Sorry but I haven't been able to reproduce the problem that changing layout is not updated with XCSoar 7.20 (from 0:00 to 0:50)

But I have been able to reproduce that changing language doesn't update menu when closing but more problematic it doesn't update info box language (from 1:10 to end). Except if XCSoar is closed and reopened

https://easyupload.io/hnbpm5

closed 2023-05-09 by MaxKellermann