XCSoar/XCSoar #708: SkyLines' "Show nearby traffic" option not persistent and not working

elgandoz, 2021-09-05:
XCSoar versions having and not having the problem
-------------------------------------------------
Since its introduction (somewhere after **6.7.6** but before **6.8**) to the current release (**7.17**)

Expected behavior
-----------------
When setting SkyLines' "Show nearby traffic" to _On_, I should be able to see other SkyLines account nearby.
After restarting XCSoar, the configuration should persist as is.

Actual behavior
---------------
No tracking from nearby.
Upon restart the value is back to _Off_.

Steps to reproduce the behavior
-------------------------------
Saving the value:
- Go to Config > System > Setup > Tracking.
- In the SkyLines section, set "Track friends" to _On_, the set the below "Show nearby traffic" field to _On_.
- Close and then Quit XCSoar, then start it again.
- If you go in the same dialog, now the "Show nearby traffic" is back to _Off_, while the rest is unchanged.

Show nearby traffic:
- Create a test SkyLines account, or use a friend's one. Set the tracking on a secondary phone via XCSoar / [SkyLines tracker app](https://play.google.com/store/apps/details?id=ch.luethi.skylinestracker), or any other working way.
- Verify the correct functioning of the tracker by visiting https://skylines.aero/tracking and check that your track is present and updated.
- Open your own XCSoar, and set "Track friends" and "Show nearby traffic" to _On_, and check the map: you won't find any Skylines position.
- If then you _follow/befriend_ that account from within Skylines, its position appear correctly.

Do you have any idea what may have caused this?
-----------------------------------------------
Not really, possibly the value is not set correctly and therefore is not read and saved.

Do you have an idea how to solve the issue?
-------------------------------------------
I may try look into the code, but i have no experience in C++

elgandoz, 2021-09-05:
Also it would be helpful having in the field's _Help_  how far is "nearby".

closed 2021-09-06 by MaxKellermann

elgandoz, 2021-09-07:
@MaxKellermann thanks for the fix, the option persist now, although "Show nearby traffic" doesn't actually work.

MaxKellermann, 2021-09-07:
Don't mix two things in one bug report. That's confusing.