XCSoar/XCSoar #709: Track is sometimes drawn incorrectly

scls19fr, 2021-09-05:
XCSoar versions having and not having the problem
-------------------------------------------------

XCSoar 7.16

Expected behavior
-----------------

Track should be drawn correctly... ie shouldn't have glitch

Actual behavior
---------------

Saturday August 28th, I noticed during a flight a strange glitch with track... see screenshot

[image]

Steps to reproduce the behavior
-------------------------------

Can't know how to reproduce

Do you have any idea what may have caused this?
-----------------------------------------------

I noticed you recently fixed a bug about airspace drawing in v7.17
https://github.com/XCSoar/XCSoar/releases/tag/v7.17
(can't find exactly bug number because it's not written in release notes but maybe you remember that)

but in this case the problem is that track seems to deforms compared to waypoints

Do you have an idea how to solve the issue?
-------------------------------------------

elgandoz, 2021-09-05:
@scls19fr the reason for the shift of the track is caused by the settings:

`Config > System > Map Display > Elements > Trail drift`: very likely you have the flag to _On_.
If you check the Help of that field, it says:

> Determines whether the snail trail is drifted with the wind when displayed in circling mode. Switched Off, the snail trail stays uncompensated for wind drift.

As you can see from the icon on the bottom right of the map, and from the presence of the thermal assistant, you are currently circling.
The drift is meant to be used while zoomed-in, in order to help centre and maintain the thermal.
If you don't wont this behaviour you can simply turn it off.

Regardless this is the correct behaviour and not a bug, please close the issue.

scls19fr, 2021-09-06:
Thanks @elgandoz I wasn't aware of this setting and so that's not a bug

Config / Système / Carte / Elements / Dérive compensée en cheminement is now set to Désactivé

Sorry for noise... closing issue

closed 2021-09-06 by scls19fr