XCSoar/XCSoar #712: Docker compile: "specified path differs in case from file name on disk" and other problems

elgandoz, 2021-09-06:
XCSoar versions having and not having the problem
-------------------------------------------------
I'm using XCSoar 7.17. 
The bug happens while compiling, the syntax of the `#include` hasn't changed in the last 7 years.

Expected behaviour
-----------------

- Follow the instruction to build for Android from Docker [as described in the guide](https://github.com/XCSoar/XCSoar/blob/master/ide/docker/README.md)
- Expect an Android build

Actual behaviour
---------------

- After a running the command to build for Android, I get the following errors in the output:
  ```[...]
  CXX     output/ANDROID/x86/dbg/src/Formatter/Units.o
  CXX     output/ANDROID/x86/dbg/src/Formatter/UserUnits.o
  In file included from src/Device/MultipleDevices.cpp:26:
  src/Device/Dispatcher.hpp:27:10: error: non-portable path to file '"Device/util/LineHandler.hpp"'; specified path differs in case from file name on disk [-Werror,-Wnonportable-include-path]
  #include "Device/Util/LineHandler.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          "Device/util/LineHandler.hpp"
  1 error generated.
  make[1]: *** [build/compile.mk:108: output/ANDROID/armeabi-v7a/dbg/src/Device/MultipleDevices.o] Error 1
  rm output/ANDROID/armeabi-v7a/dbg/src/Device/Driver/../dirstamp  CXX     output/ANDROID/x86/dbg/src/Formatter/HexColor.o

  make[1]: Leaving directory '/opt/xcsoar'
  make: *** [build/android.mk:267: output/ANDROID/armeabi-v7a/dbg/bin/libxcsoar.so] Error 2
  make: *** Waiting for unfinished jobs....
  [...]
    CXX     output/ANDROID/x86/dbg/src/Device/MultipleDevices.o
    CXX     output/ANDROID/x86/dbg/src/Device/device.o
  In file included from src/Device/MultipleDevices.cpp:26:
  src/Device/Dispatcher.hpp:27:10: error: non-portable path to file '"Device/util/LineHandler.hpp"'; specified path differs in case from file name on disk [-Werror,-Wnonportable-include-path]
  #include "Device/Util/LineHandler.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          "Device/util/LineHandler.hpp"
  1 error generated.
  make[1]: *** [build/compile.mk:107: output/ANDROID/x86/dbg/src/Device/MultipleDevices.o] Error 1
  make[1]: *** Waiting for unfinished jobs....
  rm output/ANDROID/x86/dbg/src/Device/Driver/../dirstamp
  make[1]: Leaving directory '/opt/xcsoar'
  make: *** [build/android.mk:267: output/ANDROID/x86/dbg/bin/libxcsoar.so] Error 2
  rm output/host/tools/GenerateSineTables/../dirstampmake: unlink: output/host/tools/GenerateSineTables/../dirstamp: Not a directory
  output/host/tools/GenerateSineTables.o
  ```
- After replacing the include with a lowecase folder name, the build progresses, but I hit another issue:
  ```
  [...]
    CXX     output/ANDROID/armeabi-v7a/dbg/src/ui/canvas/android/Bitmap.o
    CXX     output/ANDROID/armeabi-v7a/dbg/src/ui/canvas/android/Font.o
  src/ui/canvas/android/Bitmap.cpp:31:10: fatal error: 'android_drawable.h' file not found
  #include "android_drawable.h"
          ^~~~~~~~~~~~~~~~~~~~
  1 error generated.
  make[1]: *** [build/compile.mk:108: output/ANDROID/armeabi-v7a/dbg/src/ui/canvas/android/Bitmap.o] Error 1
  make[1]: *** Waiting for unfinished jobs....
  rm output/ANDROID/armeabi-v7a/dbg/src/Device/Driver/../dirstamp
  make[1]: Leaving directory '/opt/xcsoar'
  make: *** [build/android.mk:267: output/ANDROID/armeabi-v7a/dbg/bin/libxcsoar.so] 
  ``` 
  In this case I actually cannot find such file.
- If I remove the `#include "android_drawable.h"`, the build progresses till it requires that library: `undeclared identifier 'DrawableNames'` where it then stops.

Steps to reproduce the behaviour
-------------------------------
On MacOS 11.5.2 BigSur and using Docker 4.0.0.12:
```
❯ git clone https://github.com/XCSoar/XCSoar.git
❯ cd XCSoar
❯ docker run \
    --mount type=bind,source="$(pwd)",target=/opt/xcsoar \
    -it ghcr.io/xcsoar/xcsoar/xcsoar-build:latest xcsoar-compile ANDROID
```

Do you have any idea what may have caused this?
-----------------------------------------------
Unfortunately not.
I have no idea why it cannot find `android_drawable.h` or why the case changes.
It could be some build configuration change, maybe due Docker version or MacOS.

Do you have an idea how to solve the issue?
-------------------------------------------
Unfortunately not, but hopefully @lordfolken could give me a hint.

elgandoz, 2021-09-06:
Likely related: 
- https://stackoverflow.com/questions/60225719/how-to-suppress-specific-compiler-warnings-in-xcode11
- https://stackoverflow.com/questions/43067017/non-portable-path-to-file-file-h-specified-path-differs-in-case-from-file-na/43067018#43067018
- https://github.com/gibiansky/IHaskell/issues/942

I just found out that my FileSystem is Case Insensitive due to a recent re-format. Although a [change of FileSystem could sort this](https://support.apple.com/en-au/guide/disk-utility/dsku19ed921c/mac#dsku127e6e61), is there any other way around?

lordfolken, 2021-09-06:
We currently depend on casesensivity, sorry.

elgandoz, 2021-09-06:
For future reference, I solved by creating a new case-sensitive volume (Case-insensitive is the _default_, by the way).

With MacOS "Disk Utility" > Right Click on the main volume > Add new APFS volume... > Select "APFS (Case-sensitive)" > Add and do not assign any space (dynamically allocated).

I then recloned the repo there and run the same commands, everything worked.

closed 2021-09-06 by elgandoz

lastant, 2023-07-13:
I solved this by adding ` -Wno-nonportable-include-path` to ALL_CXXFLAGS and ALL_CPPFLAGS, ALL_CFLAGS (just in case) in the file build/flags.mk lines 64-66