XCSoar/XCSoar #715: User defined buttons added via .xci file no longer working.

7678514, 2021-09-07:
XCSoar versions having and not having the problem
-------------------------------------------------
7.16 and 7.17 not working for Win 32 & Win 64
7.15 and earlier working for Windows
7.16 Linux working as expected
7.17 Android working as expected
Don’t know if it impacts other devices.

Expected behavior
-----------------
The input event file (extension .xci)  allows additional user defined buttons to be added to the XCSoar menus. Pressing the additional buttons should execute whatever has been defined for each button.

Actual behavior
---------------
All additional xci buttons are showing on all the different menu screens where they should be, however, all xci buttons are greyed out and not executable.

Steps to reproduce the behavior
-------------------------------
Loading any xci file should reproduce the behaviour. The .xci file that I use is nothing out of the ordinary and has not been changed recently. The same file works as expected on Linux and Android.

Do you have any idea what may have caused this?
-----------------------------------------------
Appears that a bug has been introduced in 7.16 which possibly only impacts Windows.

Do you have an idea how to solve the issue?
-------------------------------------------
No

MaxKellermann, 2021-09-10:
Upload a minimal sample file demonstrating your problem.

7678514, 2021-09-11:
Attached are screens shots to illustrate problem. The 2 buttons (top right) are additional buttons added to the main screen using the xci file. These buttons are greyed out and not selectable. I have tried replacing these buttons with different functions, but it is the same irrespective what I tried.

Also attached is a stripped-down version of the xci file that I have been using for >2 years. I have removed additional menus and buttons to make sure there was no conflict with other things I had added.

[7678514 additional files.zip](https://github.com/XCSoar/XCSoar/files/7147155/7678514.additional.files.zip)

closed 2021-10-01 by MaxKellermann