XCSoar/XCSoar #716: FTP and telnet server on Kobo mini not working

istaveren, 2021-09-08:
XCSoar versions having and not having the problem
-------------------------------------------------
With problem 7.14, 7.15, 7.16
Without the problem 6.8.10

Expected behavior
-----------------
When I connect the Kobo to WIFI 
And check I get an IP address
And I click on 'FTP server'
Then I should be able to connect with FTP to the IP address shown

Actual behavior
---------------
When I connect the Kobo to WIFI 
And check I get an IP address
And I click on 'FTP server'
Then I can ping the Kobo on the IP address 
And I can't FTP to it.

`$ ping 192.168.178.90
PING 192.168.178.90 (192.168.178.90) 56(84) bytes of data.
64 bytes from 192.168.178.90: icmp_seq=1 ttl=64 time=100 ms
64 bytes from 192.168.178.90: icmp_seq=2 ttl=64 time=13.7 ms
^C
--- 192.168.178.90 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 13.662/56.834/100.007/43.172 ms
$ ftp 192.168.178.90
ftp: connect: Connection refused
`

Steps to reproduce the behavior
-------------------------------
When I connect the Kobo to WIFI 
And check I get an IP address
And I click on 'FTP server'
Then I go to a terminal on a PC
And I ftp <to IP address>

Do you have any idea what may have caused this?
-----------------------------------------------

No.

Do you have an idea how to solve the issue?
-------------------------------------------

It would be handy if you get some feedback if you click on the FTP or Telnet server buttons.
If it works fine you get UP and running connect to IP address for example.

If it fails show some log with the failure?

As I currently am expecting that the ftp server process is failing, but there is no way to find out.

MaxKellermann, 2021-09-10:
Can't reproduce. Please find the most recent version that still works, and the oldest version that does not work. You omitted lots of versions.

istaveren, 2021-09-12:
Bought a new Kobo and the issue is gone.
I guess something wrong with the SD card.

Thanks anyway.

closed 2021-09-12 by istaveren