XCSoar/XCSoar #719: CTR- Airspace not displayed

fraktal-sb, 2021-09-10:
XCSoar versions having and not having the problem
-------------------------------------------------

6.8.17 / 7.16

Expected behavior
-----------------

all airspaces (as configured) should be displayed

Actual behavior
---------------

missing airspace (s?)

Steps to reproduce the behavior
-------------------------------

- use data from https://www.lvzc.be/index.php/luchtruim/luchtruim-4 and 2021 in 7.16
- compare with 6.8.17
- see screenshots
- tested also with setup show all airspaces

[image]
[image]

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

MaxKellermann, 2021-09-10:
XCSoar 7.16 is outdated and unsupported. Always update to the latest version before reporting a bug.
Probably a duplicate of https://github.com/XCSoar/XCSoar/issues/702

fraktal-sb, 2021-09-10:
sorry, my fault, tested with with 7.17. CTR still missing

fraktal-sb, 2021-09-11:
tested also with recent version (7.18) and test-airspace file (containing only airspace Luxembourg- CTR) and old version 6.8.17 with same result: CTR Luxembourg not displayed in recent version; airspace file attached. Test under Linux/Debian did also not display the airspace. Issue should be openend again.
[CTR_TEST.txt](https://github.com/XCSoar/XCSoar/files/7147295/CTR_TEST.txt)

August2111, 2021-09-16:
The problem has nothing to do with the CTR, every other CTR works as expected: But the Arc line ('DA') wouldn't be parsed correctly - and so this airspace 'contains' only one line - and so it is removed from the view( and/or list). 
This happens in all airspace classes, not in CTR only...
Maybe next week I could solve this problem

closed 2021-09-16 by MaxKellermann

fraktal-sb, 2021-09-19:
issue should be opened again.

reason:

Airspace of type "Luxembourg" (see file CTR-TEST, or  https://www.lvzc.be/index.php/luchtruim/luchtruim-4) ist not accepted. See screenshot with error message. The problem did not exists in 6.8.17. When accepting with "OK" and closing XCSoar next time XCSoar is not usable. The program goes directly back to the splash screen. Only removing the airspace file in the "Airspace" directory helps. This was testet in the UNIX as well in the KOBO version
[image]

MaxKellermann, 2021-09-19:
Different bug, new bug report. Don't cause confusion by mixing different issues in one bug report. that'll only waste everybody's time.