XCSoar/XCSoar #720: Display issue number and PR number in NEWS.TXT

scls19fr, 2021-09-11:
Hello,

For easier XCSoar development understanding it will be nice if each (new) entry in [NEWS.TXT](https://github.com/XCSoar/XCSoar/blob/master/NEWS.txt) had a reference to an issue number and/or a pull request number

issue #xyz  PR #abc
issue #xyz commit abcdefg

The minimal effort could be to only reference issue number. So using that number it will be possible to know how it have been fixed.

Kind regards

lordfolken, 2021-09-12:
There are some automations in github actions that take issues and PR titles and add them to the changelog. Care to dive into the topic?

MaxKellermann, 2021-09-12:
GH Actions: please no, I don't want automated commits made by bots.

And I don't even agree that we need issue numbers in the changelog. If you want to know the issue number, you can always use `git blame`: https://github.com/XCSoar/XCSoar/blame/master/NEWS.txt
.. and then click on the commit, for example the "TCP Client: fix crash bug"
.. which leads to https://github.com/XCSoar/XCSoar/commit/c207c7e798a1186766782b4be4215f23c54c57b9 showing `Closes #710`.

Or do:
```
git log v7.16.. --grep=Closes
```

Having commit ids in the changelog is impossible, because that would imply knowing the commit id before doing the commit. Unless you add the commit ids later in another commit, which would modify the NEWS file again, which breaks `git blame` for no reason. This would be some sort of crippled `git blame` for people who don't know about `git blame`.

The NEWS file is for users, not for developers. Developers know how to use `git log`, `git blame` etc.

closed 2021-09-15 by scls19fr