XCSoar/XCSoar #721: Cannot declare task to IMI Erixx logger via Bluetooth (HC-05)

DanD222, 2021-09-17:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?
XCSoar v7.18

Which XCSoar version is the last one you know that did not show the problem?
The feature works on XCSoar v6.8.10-MacOSX

Expected behavior
-----------------

The task is successfully declared.

Actual behavior
---------------
A window pops up with the message: "Error occured. Task NOT declared!"

Steps to reproduce the behavior
-------------------------------
- Connect Erixx logger to IMI Connect program via USB cable and set NMEA speed to 9600
- Set Bluetooth HC-05 module baud rate to 9600
- Feed data from IMI Erixx to HC-05 via RS-232 to TTL converter
- Pair HC-05 module with Android phone
- Run XCSoar, navigate to Config / Devices, set up paired HC-05 module, select driver "IMI ERIXX"
- Verify that XCSoar is receiving Baro and GPS data from the logger
- Create or load a task, navigate to Manage / Declare, confirm declaration to IMI ERIXX

Do you have any idea what may have caused this?
-----------------------------------------------
I would guess it has something to do with the baud rate setting during the declaration - I can successfully declare to the Erixx logger connected to a computer running XCSoar v6.8.10-MacOSX via Bluetooth.

I notice that navigating to Config / Devices in XCSoar v6.8.10-MacOSX and clicking on Edit shows a Baud rate window below the Port window. I can then select 9600 as the baud rate for the logger.

Do you have an idea how to solve the issue?
-------------------------------------------
I can lend a device for testing, should it help.

MaxKellermann, 2021-09-17:
Please test the versions between and find out which was really the first broken one. And don't compare macOS builds with Android - it might be helpful to know when the same version has different behavior on different devices, but for comparing versions, it is important to have the same environment.

DanD222, 2021-09-17:
OK, I’ve tried Android v6.5.1, v6.7, v6.8.10 and v6.8.13 from http://download.xcsoar.org/releases/ and wasn’t successful on either of them unfortunately. The logger I have has the firmware updated for COMM_MAX_PAYLOAD_SIZE of 2048 as changed in 458ec04ab8a6b2cf1df3fcfed03ce3e960c5fbed, but I don't know whether this could be a factor.

MaxKellermann, 2021-09-17:
So ... this works on macOS with version 6.8.10 **without** your 2 kB fix, but version 6.8.13 **with** your fix doesn't work on Android?

And your test demonstrated that it was never possible on Android to declare a task?

DanD222, 2021-09-17:
Pretty much yes, unfortunately, at least over Bluetooth.

MaxKellermann, 2021-09-20:
I had a look at the IMI driver code in XCSoar, and it's pretty bad code which needs a rewrite. (It was written by a developer who has left the XCSoar project long ago.)
... but that can't be done without having a device. Before you send yours over to me (my address is on the imprint of the XCSoar website), maybe ask the manufacturer to donate one for XCSoar development, so we can ensure long-termin compatibility.

Turbo87, 2021-09-29:
@MaxKellermann it worked well enough and I'm pretty sure you reviewed it back then 🤷‍♂️

MaxKellermann, 2021-09-30:
No, I did not review this particular piece of code, it's the one PR I did not review, after all the criticism for being too nitpicky (being nitpicky is good, after all, and I should not have merged this code). The coding style is "strange" and its use of global variables is clearly not acceptable.
If it ever worked, this must be by chance, not by design.
For example, this piece of code is utterly broken:
https://github.com/XCSoar/XCSoar/blob/4b7f76e90eaa4238483354df18e9d3e7c97e8c36/src/Device/Driver/IMI/Protocol/Communication.cpp#L95-L100
It waits until at least one byte was received, and if there's not a complete message header, it will fail. The problem is that it doesn't even try to handle partial reads. If this read contains a part of the header, it will fail to parse it, discard the whole buffer, and issue a new read, which will contain (some of) the rest of the header, but that is doomed to fail again.
This "just happens" to work if the first read just happens to read the whole header (which is very unlikely). It failed 100% of the time in all of my attempts with @DanD222's ERIXX.

MaxKellermann, 2021-09-30:
> The problem is that it doesn't even try to handle partial reads.

I was wrong about this part. It does handle partial reads, but in an extremely obscure way. There is a global buffer variable in MessageParser.cpp which gets filled incrementally by the obscure state machine in Parse(), in a very clumsy way.

There were two major reasons for this bug: 1. the timeouts were only a few hundred milliseconds, which is way too short; 2. the driver fails silently (without any error message) if the port implementation doesn't know the baud rate:
https://github.com/XCSoar/XCSoar/blob/9a83833e1caf6062a8c52b07548955ce42385f12/src/Device/Driver/IMI/Protocol/Protocol.cpp#L71-L73

I tried removing the `STARTCONFIG` command (which requires transmitting the baud rate), but then the ERIXX stopped replying completely. This is a badly designed protocol, because there is no reason to require knowing the baud rate; the ERIXX could simply keep on using the current baud rate. I blame IMI for designing such a bad protocol.

closed 2021-09-30 by MaxKellermann